iNavFlight/inav #5859: Fixed wing: How to Tune minor Oscillations on roll? (could have been done with D-Term)?

Pairan, 2020-06-16:
Currently the Wing flies good, just some minor oscillations on the roll axis are annoying me.

Currently installed v2.4.1 - but this is surely the same as for 2.5.x

`
fw_p_roll = 9
fw_i_roll = 1
fw_ff_roll = 85
`
As the D-term has been removed from the cli as well as from the configurator, I'm a bit stuck on how to fix those oscillations.

How can I improve on that? ... raising FF?

digitalentity, 2020-06-16:
Can you please share:

1.  Dump
2. Blackbox log
3. Description of your wing

Pairan, 2020-06-16:
I will have to make a log on next flight. Never worked with logs before- anything I should know/need to have checked?

The wing is a Reptile S1100 Harrier with two motors

So far I can provide a DIFF or an outtake from that with the related settings.  I you prefer the full dump then let me know. I will provide that later. 

The Diff
# INAV/MATEKF405SE 2.4.0 Feb 11 2020 / 22:58:10 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# mixer
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1
smix 4 5 8 100 0 -1
smix 5 6 9 100 0 -1
smix 6 7 10 100 0 -1

# servo
servo 3 1000 2000 1560 -80
servo 4 1000 2000 1450 80

# feature
feature -SOFTSERIAL
feature BAT_PROF_AUTOSWITCH
feature MOTOR_STOP
feature GPS
feature LED_STRIP
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH

# aux
aux 0 0 1 1800 2100
aux 1 2 0 1400 1800
aux 2 3 0 1175 1225
aux 3 9 0 1150 1250
aux 4 8 0 1075 1175
aux 5 35 0 1800 2100
aux 6 27 3 1425 1600
aux 7 28 3 1800 2100
aux 8 11 5 1750 2100
aux 9 18 0 1075 1175
aux 10 32 6 1050 1325
aux 11 33 6 1325 1800
aux 12 34 6 1800 2100

# master
set acc_hardware = MPU6000
set acczero_x = 108
set acczero_y = -41
set acczero_z = -586
set accgain_y = 4089
set accgain_z = 4034
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 12
set serialrx_provider = CRSF
set motor_pwm_rate = 1000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set align_board_pitch = 30
set model_preview_type = 8
set small_angle = 180
set imu_acc_ignore_rate = 10
set disarm_kill_switch = OFF
set applied_defaults = 1
set gps_ublox_use_galileo = ON
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = FS_ONLY
set nav_rth_alt_mode = AT_LEAST_LINEAR_DESCENT
set nav_rth_altitude = 10000
set nav_fw_cruise_thr = 1450
set nav_fw_launch_detect_time = 20
set nav_fw_launch_thr = 1600
set nav_fw_launch_timeout = 20000
set nav_fw_launch_climb_angle = 20

# profile
profile 1

set fw_p_pitch = 9
set fw_i_pitch = 20
set fw_ff_pitch = 91
set fw_p_roll = 9
set fw_i_roll = 1
set fw_ff_roll = 85
set fw_i_yaw = 14
set fw_ff_yaw = 63
set dterm_setpoint_weight =  0.000
set nav_fw_pos_z_p = 35
set nav_fw_pos_z_d = 15
set rc_expo = 60
set rc_yaw_expo = 40
set manual_rc_expo = 60
set manual_rc_yaw_expo = 40
set manual_pitch_rate = 80

schnupperm, 2020-06-18:
I have the same problem (Dart 250G, under 250 g).
Is it save to set pid_type=PID on a fixed wing, to get the D-term back?

digitalentity, 2020-06-19:
It's generally safe to change the `pid_type` to `PID`, but you'll have to figure out proper P-gain. Generally, oscillations could be caused by any of the PIFF terms. 

@Pairan based on your dump you may want to have lower FF and higher I and P for roll. Generally FF is is a direct connection to you stick in ACRO (think "manual mode"), while P and FF offer actual stabilization against plane movement. Try lowering FF considerably and raising P - should help get oscillations under control. What are you `roll_rate` and `pitch_rate` parameters set to?

Pairan, 2020-06-19:
@digitalentity 
I will try that- in the moment I wrote this, I was a bit clueless on how to get hold on it. But digging a bit in the wiki I found to raise P and I a bit.
Lowering FF and raising P will get a shot on next flight- as well as a log if 2GB SD will be sufficent for such a task.

I'm a cruiser making these videos while passing over the landscapes so usually I stick in HOR or CRZ. Reduced Pitch and Roll rate a lot ... but I need to take a look at the settings, when im back on the Harrier. By that time I will report back in.
Thanks for your support ;) Greetings goes out to you in Switzerland!

digitalentity, 2020-06-19:
I have an S1100 on the way - fingers crossed that would be my main HD-wing, so I guess I'll have the same issue as you )

Pairan, 2020-06-19:
It's more a quality detail than an issue ... but to calm a wing down so that 4K footage can go smooth is a lovely thing to do!
If you need anything/infos around that wing, let me know ... I guess I can tell you a lot. Currently running 50min on a flight

tiriad, 2020-06-19:
Try to increase i and decrease p?
the following video explains it in detail https://youtu.be/A45vc4OihgY?t=584

Pairan, 2020-06-19:
I will be honest: I didn't like the way the wing got tuned in Pawels video. My wings usually fly  pretty smooth. But the tuning is (/ my settings are) far from what Pawel suggests. Nothing to complain on Pawel's work, so don't get me wrong. 

The main problem seams, that everybody has his own way to tune! 

- The Wiki says: raise P for stability!
- in this Thread DigitalEntity suggests Raising P and lowering FF
- you suggest lowering P and raising I 
... 

I try to stick to what iNav offers/suggests - make the wing fly and perform an autotune, which I have had good results with
Then i start tweaking on it as the final result isn't far from the autotune.

My Problem is, that I look for a solution which is "up-to-date"  (P-I-FF) and like what "raising D a bit" would have achieved

dragnea, 2020-07-04:
> This is surely a SUPPORT question!
> Currently the Wing flies good, just some minor oscillations on the roll axis are annoying me.
> 
> Currently installed v2.4.1 - but this is surely the same as for 2.5.x
> 
> fw_p_roll = 9
> fw_i_roll = 1
> fw_ff_roll = 85
> 
> As the D-term has been removed from the cli as well as from the configurator, I'm a bit stuck on how to fix those oscillations.
> 
> How can I improve on that? ... raising FF?

Hey.
Lower p and raise i to benefit of stabilisation in air mode. If you want to cruise, you’re using cruise and/or implicit angle mode, so imho try to lower the angle strength but after you put some i term... also, try to add some Adjustments. It will make your life easier tuning in the air... https://youtu.be/KoJmKmrHE50

Pairan, 2020-07-08:
I will give it a try!

The adjustments was something I tried out some time ago. But I found it to be a hassle to configure for a temporary setup. Ended up with the idea, if I would do so, I would copy the model (in the Transmitter) and program the adjustments into the copy! Then, when the adjustments have been made, simply dispose the copy.

Thanks for your reply

closed 2020-09-20 by stale[bot]