iNavFlight/inav #5860: msc mode does not work anymore

diablol, 2020-06-16:
Hello.
Since 2.5 msc does not work anymore.  I tried back and forth with 2.4. It works with 2.4
With a Matek F722 on Win 10 and Inav 2.5 the storage mode does not work. 
pls tell me if I am wrong here or can provide more information.

Mateyhv, 2020-06-17:
Hello, try installing 2.5.1 first wich is the current stable release.

diablol, 2020-06-17:
Hi, I did. I tried both Inav 2.5.0 and 2.5.1 with configurator 2.5

diablol, 2020-06-17:
Even though I never had problems with earlier version it is properly a (local) computer problem. 
because it works with Android OTG storage...

breadoven, 2020-06-17:
Works for me as before on Windows 10.

Torchi, 2020-09-24:
It seems I have the same problem of diablol.

These are my SW versions:

INAV version: 2.5.2
FLIGHT CONTROLLER Matek F722-SE
Windows 10 Pro 64 bit (updated to June 2020)

I follow the instructions:

Enter the CLI
Enter the CLI command msc ; the FC will reboot
Close the CLI tool (cliterm, configurator etc.)
Wait for the device to be recognised as USB storage device by the operating system (may take some time, 10-15 seconds perhaps).
.....

nothing happens in Windows: no new unit is mounted. 

Using CLI via Speedy Bee on Android, instead, the SD can be mounted.

iblackymb, 2020-10-18:
Same behaviour here, the board reboots, but windows didnt recognize it.

MrMap, 2020-10-25:
Same problem here using Matek F765. 
If it shows up at all in Windows then it is not recognized as a Mass Storage Unit but rather as an unknown. 
Tried on Win 10 and 7 - same. 
I followed the steps in the documentation.

Version info:
# version
# INAV/MATEKF765 2.5.2 Aug  4 2020 / 10:30:32 (faaedc7cb)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

kniuk, 2020-11-03:
same for me: doesn't work on Windows 10, does work on Android.

b14ckyy, 2020-11-14:
Current Behavior
If MSC mode is enabled via CLI on the F722-WPX Flight Controller, the FC will freeze and not show mass storage device on PC

Steps to Reproduce
Flash INAV 2.5.1 from Matek with SD and baro drivers or INAV 2.6 Alpha 2
go to CLI and type "msc" and hit enter
Windows will play disconnect and connect sound and configurator shows "Rebooting" message
COM-Port still shown as available in configurator and device manager and no mass storage device will be shown
Expected behavior
COM-Port should disappear and a mass storage device should pop up

Suggested solution(s)
Bugfix

Additional context
Tested on 2 different Windows 10 PCs. All drivers working for configuration and flashing
Tested with a F405 Wing FC and same INAV 2.6 Alpha2 Build and works fine there
Same issue reported by another user on Facebook with same FC and 2 different PCs
Possible also F722-PX affected and @MrD-RC will test that on his

FC Board name and vendor: Mateksys F722-WPX Wing

INAV version string:
INAV/MATEKF722WPX 2.6.0 Nov 6 2020 / 14:17:31 (d0046f3)
GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

MrD-RC, 2020-11-15:
> Possible also F722-PX affected and @MrD-RC will test that on his

It worked fine for me with the F722-PX on both 2.5.1 and a 2.6 build from Thursday evening.

MrMap, 2020-12-07:
Problem remains on iNav 2.6.0, FYI.

b14ckyy, 2020-12-08:
Thanks for the reminder. Yes also not working for me with the Release 2.6. 2 different PCs tested with 2 different Windows 10 builds. F405 Wing MSC mode just works fine on 2.5.2 and 2.6 on both PCs.

MrMap, 2020-12-08:
Meanwhile, while our equipment is subject to this issue, the option is to eject the SD micro card from the FC and insert it into our PC card reader. In my case the process of ejecting the card and even worse, inserting it again, was quite frustrating due to 90 degree slot angle in a tight space. So I designed a simple tool that made it possible to remove / insert SD card without actually unscrew and lift the entire FC board. 
Here is the link to the SD micro card tool stl file:
https://www.thingiverse.com/thing:4679458

b14ckyy, 2020-12-09:
@MrMap that's a really nice tool. Luckily on the F722-WPX the SD is on the Top. What's interesting though, in your case the F765 is shown as unknown device in the device manager. In the F722-WPX the device just reconnects and it is still shown as aa COM Port. But with no connection possibilities of course.

stronnag, 2021-01-03:
Reported (by @shellixyz) as working  with F722 / 2.6 release / Fedora Linux 2021-01-03. Thus any investigator will need native access to a Windows box.

skorokithakis, 2021-01-03:
`msc` works for me on Linux with a Matek F405 wing on 2.5.

b14ckyy, 2021-01-04:
> Reported (by @shellixyz) as working with F722 / 2.6 release / Fedora Linux 2021-01-03. Thus any investigator will need native access to a Windows box.

What F722? STD, Wing, HD or WPX? 
For me only the WPX does not work in MSC mode. F405 Wing works just fine. All other FCs are either BEtaflight or have no BBX.

stronnag, 2021-01-04:
AFAIK this is only a F7xx problem, F405 works on all operating systems.

b14ckyy, 2021-01-04:
> AFAIK this is only a F7xx problem, F405 works on all operating systems.

the F722-PX Quad FC with on board flash seems to work fine too

stronnag, 2021-01-04:
Pedant! I

F7xx / SDCard / Windows then

Billgyro, 2021-01-13:
I have matek F722wpx/Inav2.6/Linux Mint.  After msc SDcard appears on computer and directory structure is correct but the downloaded black box log file is the correct size but totally corrupted.  No sign of text header and MWP says it is not a black box file.  Works fine if card is removed from the FC

Billgyro, 2021-01-21:
Ignore my last post.  The FC did not like the SD card.  MSC to linux mint works fine with another brand of card.

JeremyPC, 2021-02-18:
I have this issue with my setup. I have a Matek F722WPX, with iNav 2.6.0 on two windows machines. I have not been able to test it on a linux machine yet. I tested by waiting for at least 1 minute for the 15-20 second delay that the documentation states.

I am using a SanDisk Ultra 16GB card, with the formatting tool applied as suggested at: https://github.com/iNavFlight/inav/blob/master/docs/Blackbox.md

The configurator is showing on the BlackBox page that my card has 11.0GB unavailable and 3.9GB is available, so the card is working and talking to iNav. The card is installed in the FC below the rear sticker in a dart250g, so I'd prefer to avoid needing to remove the card. There should be a few logs on it already to read.

I attach a diff dump (safehomes are configured but removed for privacy).

[dart250.txt](https://github.com/iNavFlight/inav/files/6000509/dart250.txt)

Michael-Ringstrasse, 2021-05-14:
Hello guy´s,
is this issue fixed?
I have still the same problem.
New F722-WPX with INAV 2.6.1. Windows 10 latest update, tested on to PC´s

SD card will be accept from INAV. In the chapter BLACKBOX I can see the size and freee space of the card. The same Card was working in a F405 Wing very well.  

I testet INAV 2.6.0 also without a result.

Can anybody help??

Michael

RipperGOT, 2021-07-12:
I have the same problem in Matek F722-STD

When I type the MSC command it restarts but Inav says: fail to open serial port. 

[image]

Sekilsgs2, 2021-08-21:
Any updates???? matek f722 se not working msc.

shellixyz, 2021-08-21:
Last time I checked it was working fine at least on Linux. I don't have Windows

Sekilsgs2, 2021-08-22:
> Last time I checked it was working fine at least on Linux. I don't have Windows

not working win 10 x64.

b14ckyy, 2021-08-22:
> Last time I checked it was working fine at least on Linux. I don't have Windows

anything we can give you? Logs? Something specific? It must be FC related because oither FCs work in MSC mode like F765 and F405 on the same machine. Only F722 cause problems.

shellixyz, 2021-08-22:
There is not much you can do. A dev would need to reproduce the issue and take a look at what's causing it. I have one F722. I might take a look whether I can reproduce. Is the issue with SD cards or with dataflash chips or both ?

b14ckyy, 2021-08-22:
> There is not much you can do. A dev would need to reproduce the issue and take a look at what's causing it. I have one F722. I might take a look whether I can reproduce. Is the issue with SD cards or with dataflash chips or both ?

I only have one with SD Card. Darren Lines tested with a F722-PX with on board flash and it seemed to work

shellixyz, 2021-08-23:
Indeed doesn't work with F722 and WIN10. Even on-board flash isn't working for me. As stated before works fine on Linux though. Would be nice if someone with "Windows skills" and hardware could take a look.

b14ckyy, 2021-08-25:
@DzikuVx maybe?

Steve-A15, 2021-10-05:
I'm having the same problem with INAV 3.0 and Matek F722-SE and Windows 10. I will attempt this with one of the recommended microSD cards and see if that makes any difference.

DzikuVx, 2021-11-07:
Windows MSC was broken by https://github.com/iNavFlight/inav/pull/5475/commits/90e127c6969bb0cd8f82c3329c83f6aff7528569

DzikuVx, 2021-11-14:
Fixed in 4.0

closed 2021-11-14 by DzikuVx

SebastianKumor, 2022-09-02:
MSC command does not work on MAMBAF722_I2C with inav 5.1.0 with 5.1.0 configuration on mac. I will try windows later as well

lida2003, 2023-07-03:
It doesn't work for me also. https://github.com/iNavFlight/inav/discussions/9155