iNavFlight/inav #5863: Run away poshold with good mag, fc upside down

fremebondo, 2020-06-18:
Hi,
I just built a quadcopter with mamba f405 mounted upside down, arrow toward the front, bn880 installed on the rear of the quad with gps antenna toward the sky (BN880 label facing ground) and cables exiting to the front.

Mag is oriented and calibrated, headings are correct in setup screen, even when changing pitch.

Heading calculation seems stable to me even in flight, as you can see in bb log here.

> 
[blackbox_log_2020-06-18_200008.TXT](https://github.com/iNavFlight/inav/files/4801117/blackbox_log_2020-06-18_200008.TXT)

In this short log I flew in my garden. Quad stable in althold mode. I never changed heading during flight and calculated heading seems good to me, but as soon as I enable poshold the quad runs away. It seems diverging, no sign of feedback trying to go back to position. As my backyard is small I have to stop the quad with sticks almost immediately, but calculated target pos in bb log already should say something...if you can understand better than I can.

May an issue exists with upside down fc and poshold?
Any ideas?

On Saturday I'll try to gain altitude at flying field and leave the quad run away for a bigger distance to see if It will ever try to come back.

Thanks for the efforts spent in this great project!

fremebondo, 2020-06-19:
`

# diff

# version
# INAV/MAMBAF405 2.4.0 Jun 18 2020 / 16:34:47 (e31c9f2b2)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer
mmix 0  1.000  1.000  1.000  1.000
mmix 1  1.000  1.000 -1.000 -1.000
mmix 2  1.000 -1.000  1.000 -1.000
mmix 3  1.000 -1.000 -1.000  1.000

# servo mix
smix 0 0 29 -125 0 4
smix 1 0 29 125 0 5

# servo

# logic
logic 0 1 3 1 7 0 1500 0
logic 1 1 2 1 4 0 1700 0
logic 2 1 7 4 0 4 1 0
logic 3 1 2 2 17 0 1 0
logic 4 1 8 4 2 4 3 0
logic 5 1 12 4 4 0 0 0

# gf

# feature
feature DYNAMIC_FILTERS
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 2 115200 57600 0 115200

# aux
aux 0 0 2 1425 2075
aux 1 1 2 1425 2100
aux 2 26 2 1425 2100
aux 3 3 4 1200 2100
aux 4 9 4 1650 2100
aux 5 8 5 1425 2100
aux 6 19 3 1375 2100
aux 7 17 2 900 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# master
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 70
set gyro_notch1_hz = 170
set gyro_notch1_cutoff = 125
set gyro_notch2_hz = 85
set gyro_notch2_cutoff = 43
set dyn_notch_min_hz = 90
set acc_hardware = MPU6000
set acczero_x = 42
set acczero_y = -8
set acczero_z = -2
set accgain_x = 4109
set accgain_y = 4094
set accgain_z = 4054
set align_mag = CW90
set mag_hardware = HMC5883
set magzero_x = -19
set magzero_y = -169
set magzero_z = 17
set baro_hardware = BMP280
set pitot_hardware = NONE
set serialrx_provider = IBUS
set motor_pwm_rate = 8000
set motor_pwm_protocol = DSHOT300
set align_board_roll = 1796
set align_board_pitch = -6
set model_preview_type = 3
set disarm_kill_switch = OFF
set applied_defaults = 1
set rpm_gyro_min_hz = 90
set gps_sbas_mode = AUTO
set name = FremeDrone

# profile
profile 1

set mc_p_pitch = 80
set mc_d_pitch = 18
set mc_p_roll = 80
set mc_d_roll = 18
set mc_p_yaw = 95
set yaw_lpf_hz = 30
set dterm_notch_hz = 125
set dterm_notch_cutoff = 90
set rate_accel_limit_roll_pitch = 3600
set rate_accel_limit_yaw = 720
set heading_hold_rate_limit = 30
set roll_rate = 40
set pitch_rate = 40

# battery_profile
battery_profile 1

set vbat_warning_cell_voltage = 355
set battery_capacity = 2200
set battery_capacity_warning = 880
set battery_capacity_critical = 330

# end the command batch
batch end
`

digitalentity, 2020-06-19:
Fly away in poshold usually mean mag malfunction. Is the heading correct in the field and in all directions (North, South, East, West)?

fremebondo, 2020-06-19:
Thanks @digitalentity,  I read almost everything about MAG alignment. Today I tried rotating BN880 on z axis to have cables exitign form the rear. Of course I also rotated by 180° the configuration. The MAG reading are perfect to me. Precise on the ground in every position, and pitch, and also in the air, flying in straight line. I flew in a wide space and tryied leaving poshold ON without stick inputs: the quad starts a very large bowl (50m diameter) then go back to where it should stay with a different heading. The entity of this bowling is dependent on initial heading. I found a starting heading that minimize or cancel the problem. Everything what's happening says me that there is some mag alignment problem, but readings from configurator and Blackbox are fine to me. For this reason I suspect some kind of problem with non standard FC alignment (upside down). I planned to fly tomorrow switching the orientation of my fc back to straight.

Here I attach a log: on the very first straight line (without POSHOLD) I flew over [a](url
[blackbox_log_2020-06-19_130226.TXT](https://github.com/iNavFlight/inav/files/4804386/blackbox_log_2020-06-19_130226.TXT)
) road heading 326°.

fremebondo, 2020-06-19:
Also, I cannot find GPS signals on BBBExplorer (for example GPS_ground_course as referred in https://github.com/iNavFlight/inav/wiki/INAV-blackbox-variables)

stronnag, 2020-06-19:
There is large variation between mag and GPS CoG (assuming you fly nose first all the time).

fremebondo, 2020-06-19:
How can I view COG?

fremebondo, 2020-06-19:
No COG signal in the heading of my log, line 4, H Field I ....

fremebondo, 2020-06-19:
Today I was flying with mag aligned CW270, exactly as stated [here ](https://github.com/iNavFlight/inav/issues/4376#issuecomment-465049111) BN880, gps up, label down, cables exiting from rear, FC upside down but oriented toward the front.

fremebondo, 2020-06-19:
https://photos.app.goo.gl/wBWRRbTMkPSpNCZF9

I don't think it's support. To me, it's a bug. Flipped FC upside UP, Roll=0°, MAG CW270Flipped and the problem is gone.

fremebondo, 2020-06-19:
It seems fixed in 2.5.1, but I found nothing in release notes. Can you please point me there? I am doing this to learn something , and also undertanding release notes is quite important. Thank for your help. By the way.... How to plot COG in bbexplorer?

digitalentity, 2020-06-19:
2.5.1 didn't change anything regarding compass handling. List of fixes for 2.5.1 is minimal and mostly touching VTX and ESC telemetry. It's either config (incorrect alignment or calibration inside the building in presence of rouge magnetic fields) or environment (external magnetic fields, i.e. a large concrete building or a metal underground object)

fremebondo, 2020-06-19:
did you watched the video?
data are different between setupscreen and log blackbox

AlexAlle2020, 2020-06-21:
> Fly away in poshold usually mean mag malfunction. Is the heading correct in the field and in all directions (North, South, East, West)?

If you will, a similar problem.

https://github.com/iNavFlight/inav/issues/5835#issuecomment-646672648

closed 2020-09-05 by stale[bot]