iNavFlight/inav #5864: Big glider oscillate(1-2sec) on Roll in CRUISE mode

railbox, 2020-06-19:
This is a support topic.
## Current Behavior
The ASW28 glider (2.5m wingspan) start to oscillate on roll with NAV CRUISE and 3D CRUISE.
Tried different P and I for Roll PID but no changes. Also tried to increase nav_fw_pos_xy_p to 120.

## Desired Behavior
Smooth flying as in AIR mode. 

## Suggested Solution
Well maybe the 2.5.0 heading PID will help here but i didn`t tested it yet. 

## Additional context
INAV 2.4.0. Omnibus F4 V3, GPS: BN-220. Wing: ASW28 2.5m. Really slow (60 deg/s max) on roll.
https://youtu.be/WkpmNneknhg
[LOG00026.TXT](https://github.com/iNavFlight/inav/files/4803041/LOG00026.TXT)

Pairan, 2020-06-19:
look at #5859  ... same stuff!

railbox, 2020-06-19:
Well I don`t agree because in that ticket look like oscillations in AIR mode too. I dont have any problems with AIR mode. For me it is more heading hold issue in NAV modes. Not sure I understand fully how the CRUISE mode works.

avs2, 2020-06-19:
I don't know for sure if this is the same problem but you can have a look at the recent series by Andrew Newton on rudder on INAV. It could be that your rudder is overreacting and causing the roll.

railbox, 2020-06-19:
I saw it. I tried to play with P and I for yaw but as I understand the NAV engine use roll for heading control.

avs2, 2020-06-19:
Have a look at 'fw_reference_airspeed' and 'fw_turn_assist_yaw_gain'. If you do have a rudder problem then setting the latter to a value to lower than 1, or to zero even, may solve your problem. Andrew has a video about this as well.

railbox, 2020-06-19:
So is it possible to decrease roll gain and increase rudder gain for NAV mode correction?

avs2, 2020-06-19:
Let me say that I'm just a user, not a dev. Also, I don't know what problem you have exactly. A blackbox log or a video of the rudder could show what happens. But if it is the problem with the rudder being overactive in the nav modes, why would you want to reduce the roll gain?

railbox, 2020-06-20:
Today, had a couple of test flights: Looks like I was able to reduce this problem by lowering nav_fw_pos_xy_p to 40 after that the 3D CRUISE was flying quite nice and RTH deliver wing to starting point quite nicely. Adding P or I for YAW make it more worse.
 
But loitering is not good enough. the First 270 deg was really nice but after because of wind the NAV was over reacting and loitering when crazy. Here probably I need to play with fw_turn_assist_yaw_gain or nav_fw_pos_xy_i as I see rudder is really overreacting.
[image]

closed 2020-06-22 by railbox

hali9, 2020-06-23:
I call this phenomenon "keying".
When there is no wind set nav_fw_pos_xy_p=75 and then the navigation is great.
but when the wind is I must to lower it to 50, because the plane starts to keying.
I find that problem is in calculateVirtualPositionTarget_FW in navigation_fixedwing.c 
```
   // Limit minimum forward velocity to 1 m/s
    float trackingDistance = trackingPeriod * MAX(posControl.actualState.velXY, 100.0f);
```
it uses velXY for calculate which is speed over ground and should be speed over air, so I try:
```
float trackingDistance = trackingPeriod * MAX(posControl.actualState.velXY, pidProfile()->fixedWingReferenceAirspeed);
```
and the keying was lower but not always, so I expanded my calculations this way:
https://github.com/iNavFlight/inav/compare/master...hali9:keying-prevent
I tested it using virtual airspeed sensor (works when the wind strength and direction are relatively stable) and based on throttle calculation (works well but requires determination a maximum value).

This basically solved the problem.

railbox, 2020-06-23:
So you can consider as a BUG? Can it be added to plan for solving?