iNavFlight/inav #5865: New stabilized flight mode for fixed wing aircraft with yaw control

avs2, 2020-06-19:
## Current Behavior

In `ACRO`, all three axes are controlled independently. This may be desirably for pilot who fly acro manoeuvrers, but leads to unexpected behaviour for all other pilots. When the air plane is banked, the rudder tries to counteract the resulting turn because this is seen as uncommanded yaw. This leads to a cross-controlled air plane. The same behaviour occurs in `ANGLE` and `HORIZON`, where it is definitely not needed.

`TURN ASSIST` provides a a partial solution by automatically applying rudder and elevator when the plane is banked. However, many pilots don't want to use such a "mix", so I don't consider this a full solution. Moreover, `TURN ASSIST` requires turning parameters that you currently cannot change in the OSD menu or the configurator, only in the CLI.

Setting the yaw P and I gains to zero, or using rc yaw instead of stabilized yaw solves the problem, but at the cost of losing all yaw stabilization.

## Desired Behavior

There is a mode (perhaps called `STABILIZE` or `STAB`) that flies similar to `ACRO`, but the rudder does not counteract turns. I think this should then replace `ACRO` as the default mode because it is probably only a small minority that use INAV on fixed wing aircraft for acro flying.

## Suggested Solution

I understand that this may not be an easy problem to solve. After all, how does the flight controller know that you want to turn, or that it is a gust of wind that need to be corrected? Ideally, you'd have some sort of equation that determines the _expected_ turn rate based on the bank angle so that the flight controller can correct any additional yaw movement. If I understand correctly, `TURN ASSIST` does something like that, but not exactly the same, because it calculates the _desired_ turn rate for a nice coordinated turn. 

As a simpler solution, the yaw gains could be reduced proportional to the bank angle. For sharp bank-and-yank turns, this would momentarily disable stabilization, but keep yaw stabilized in level flight.

I am by no means an expert on this, so there may be better ways to tackle this.

## Who does this impact? Who is this for?

All pilots that fly fixed wing aircraft that have direct yaw control, either with a rudder or differential thrust.

## Additional context

This feature request is a follow-up from #5800 . See also [this](https://www.youtube.com/watch?v=RzWn-VABREc) and [this](https://www.youtube.com/watch?v=2Pp1jbMRL0I) video by Andrew Newton, and the comments on those videos. I think a lot of pilots would be happy with such a flight mode :)

digitalentity, 2020-06-20:
A workaround might be to disable yaw stabilization (freeze I and zero P) when there is pilot input on roll or yaw axis. If the pilot doesn't command anything - keep full yaw stab enabled.

mshagg, 2020-06-21:
I can understand why Devs wouldn't want to break a long established naming convention, but I think it's useful to reflect on the use of these terms in other flight controllers/receivers.

Acro mode in its current implementation is akin to what other gyro systems call a heading hold mode. The behaviour described by the OP is more akin to a traditional rate mode gyro.

The proposed solution is similar to what other ecosystems call stick priority, although it would be dialling back gains on the yaw axis in response to inputs on the roll axis.

notamedclosed, 2020-06-21:
I'd be happy to do some testing and offer input.  I'm a full scale pilot and on any larger aircraft we have an autopilot function called Yaw Damper.  Its primary role is to eliminate an effect called dutch roll, but it functions secondary to reduce the effect of turbulence, and adverse yaw.

This is the exact function needed to turn a inav plane with rudder into a true "fly-by-wire" machine.  The pitch and roll axis lock of ACRO with a pilot induced yaw aware gyro response on the rudder.

I've got a plane with a Matek F405-Wing for blackbox logging and it has a rear camera that can see the tail and rudder surfaces.  It's actually an A-tail so it's a rudder-elevator but still should be able to directly observe tail surfaces.

Mateyhv, 2020-06-22:
There is no easy way to distinguish between pilot input or a wind gust in ```ACRO``` since stick is no longer applied once you reach the desried bank angle. To be honest I don't see the need of yaw stabilization in ```ACRO``` mode at all. In a plane with a tail the vertical sabilizer is what gives the yaw stabilization aerodynamicaly, in a turn its a pilots responsability to coordinate aplying rudder. And if he goes for some sport acrobatic flying, last thing he needs is an algorithm not aware of whats going on messing with the tail.

```ANGLE``` is a different story because there is a constant pilot input during a maneouver so that the system knows when it is a commanded response. The maneouvers are also quite limited by max angle which makes things simpler. I think its enough to dynamicaly reduce or even block ```I``` therm above certain bank degree to stop the rudder counteracting the turn.

notamedclosed, 2020-06-23:
> There is no easy way to distinguish between pilot input or a wind gust in `ACRO` since stick is no longer applied once you reach the desried bank angle. To be honest I don't see the need of yaw [...]

Then why bother with stabilization at all?  We can build aircraft with dynamic stability.  Just add a bit of dihedral and boom, your roll is stabilized aerodynamically.  Who needs iNav?  Except that active stabilization by a computer is faster and better.  It's the same with Yaw.

As proposed yaw stabilization works the same way as pitch and roll stabilization in Acro, with the addition it needs to be able to understand the effect of a banked aircraft and determine if the aircraft's movement about the vertical axis is as desired.  It will dampen turbulence and uncommanded movement while not fighting an aircraft turning.  It absolutely has value, and as I've said, in manned aircraft equipped with such a system it is almost always on, even if the rest of the autopilot is off.  In terms of an FPV craft it will lead to smoother experience through the goggles. Yaw waggle is a common affection of most flying wings, and even airplanes with tails/rudder are not immune.  With this functionality I'd be inclined to maybe try something like drag rudders too on a wing.

I don't know anyone flying iNav with a sport/3d aerobatic aircraft.  However, if they were they wouldn't be happy with ACRO mode as it is now either.  I've flown very maneuverable parkjets with an FC that had a rate/acro mode and experienced the same issue.  Yaw is fundamentally linked to bank an airplane, it can not be decoupled like a multicopter.  During normal flight when you roll into a turn you only need to apply rudder when using the ailerons because of adverse yaw.  Once you reach your desired bank angle and return the ailerons to neutral the rudder should be centered.  Now as you continue your turn iNAV in ACRO mode sees a uncommanded yaw error accumulating and it will use the rudder to resist it.  Most iNav aircraft either don't even have a rudder (wing), or it isn't very strong.  Try that in a 3d airplane with a massive rudder and I think we would see some interesting effects.  I certainly saw some funky things in my parkjet.  So the pilot is forced to input the "incorrect" use of rudder during the steady state turn.  But it's hard to say how much rudder since we have no slip indicator. So you are just guessing.  I've seen this on my own iNav aircraft, thankfully the rudder is fairly anemic and being a little off isn't really a problem.  It would be better though if a flight controller didn't ask you to make an input that you wouldn't need if it didn't have a flight controller.

It's actually somewhat simple to decipher the required yaw rate in a turn.  It's a calculation from airspeed and bank angle, (G * Tan(Bank Angle)) / V.  There is a "correct" amount of vertical axis angular change for a given speed and bank angle to fly coordinated and any movement that occurs outside of that parameter should be corrected.  iNav already does this with TURN ASSIST mode.  It's less then perfect because most of us don't have an airspeed sensor so it uses a preconfigured value if you don't have one.  

Take something similar to that code, and add it to a new version of ACRO mode where it fights uncommanded movement about the yaw axis, but understands that a 30 degree bank at 40 mph will have a rate of turn of 18° per second and that only deviations from that should be considered error.  It won't be accurate without an airspeed sensor, but it should be ok as long as you configure that speed to a common speed range.

avs2, 2020-08-10:
I have done some further experimentation with `TURN ASSIST` on top of `ACRO` by putting it on a switch so that I can easily see effect. On an airplane with a very effective rudder (Volantex Ranger G2) I notice that when I bank the plane the rudder tips it into a spiral. On a ZOHD Talon GT, which has a much less effective V-tail rudder, this is less pronounced but still noticeable. I suspect the problem arises because INAV doesn't know about/accounts for the roll effect of rudder that most planes have. It would be nice if it was possible to set the roll rate on yaw input or something

avs2, 2020-09-23:
> A workaround might be to disable yaw stabilization (freeze I and zero P) when there is pilot input on roll or yaw axis. If the pilot doesn't command anything - keep full yaw stab enabled.

I think yaw stabilization should be disabled when the plane is banked, not when there is no stick input. When making a turn in acro you give some roll input intil the plane is at the desired bank angle, but after that point the plane will make a turn without any more pilot input.

digitalentity, 2020-09-24:
> Take something similar to that code, and add it to a new version of ACRO mode where it fights uncommanded movement about the yaw axis, but understands that a 30 degree bank at 40 mph will have a [...]

This is exactly how ACRO + TURN ASSIST behave.

> I think yaw stabilization should be disabled when the plane is banked, not when there is no stick input. When making a turn in acro you give some roll input intil the plane is at the desired bank angle, but after that point the plane will make a turn without any more pilot input.

"Banked" is not quantifiable. Is 10 degree == "banked"? Is 5 degree banked? Is 0.5 degree banked? From stabilization perspective anything that is not exactly zero is "banked". There is no "one size fits all" solution here, so I'm sticking to the TURN ASSIST logic which guesstimates the desired turn rate and tries to enforce it.

closed 2020-09-24 by digitalentity

avs2, 2020-09-24:
> "Banked" is not quantifiable. Is 10 degree == "banked"? Is 5 degree banked? Is 0.5 degree banked? From stabilization perspective anything that is not exactly zero is "banked". There is no "one size [...]

I understand that, but implementing an cutoff point such as 5 or 10 degrees roll could work. My problem with TURN ASSIST is that on all of my planes it just doesn't fly _nice_. It always feels like there is someone else adding extra roll input or playing with the pitch which makes it very awkward to fly. While TURN ASSIST may be a theoretical solution to this issue, it just doesn't work well enough in many cases.

I understand that most people here fly wings, but I read a lot of complaints about INAV and rudder on YT and FB so I think it would be worthwhile to address these issues.

digitalentity, 2020-09-24:
Again, there is no "one size fits all" solution, therefore we're sticking to one that at least has laws of physics backing it.

Disabling stabilization on Yaw is trivial - zero out P, I and D and leave only FF. You'll still have stick control, but no yaw stabilization.

avsaase, 2020-12-07:
> There is no easy way to distinguish between pilot input or a wind gust in `ACRO` since stick is no longer applied once you reach the desried bank angle. To be honest I don't see the need of yaw [...]

I am playing around with something like this and I wonder if this should apply to pitch as well. It may not be that noticeable as with yaw, but making a banked turn also induces a pitch rotation in the plane reference frame. WDYT?

paralin, 2021-05-17:
> Disabling stabilization on Yaw is trivial - zero out P, I and D and leave only FF. You'll still have stick control, but no yaw stabilization.

Never thought of this, I hope it helps with yaw instability causing roll instability on my micro flying wings.

avsaase, 2021-05-18:
@paralin wing normally don't have yaw control, so then making these changes has no effect.

Mateyhv, 2021-05-18:
> > Disabling stabilization on Yaw is trivial - zero out P, I and D and leave only FF. You'll still have stick control, but no yaw stabilization.
> 
> Never thought of this, I hope it helps with yaw instability causing roll instability on my micro flying wings.

Beware, thats true for PIFF controller but not for PID controler that is now selectable for planes.

UltraFly, 2026-09-06:
I'm investigating in my INAV branch, the possibility of adding a "Gyro Assist" feature that could be enabled while flying in MANUAL mode. If the results are promising, I'll submit a pull request.

The idea would be to provide gyro stabilization while retaining the direct stick-to-control-surface behavior of MANUAL mode, somewhat analogous to the stabilization systems commonly used in traditional RC aircraft, such as Spektrum AS3X, FrSky stabilization receivers, PowerBox/iGyro, etc.

The motivation is that INAV's ACRO mode is fundamentally a rotation-rate command mode: stick position represents a requested angular rate, and the controller moves the surfaces as necessary to achieve that rate.

That control model works very well for multicopters and helicopters, and can certainly also be used on fixed-wing aircraft, but it does not reproduce the traditional control feel expected by many pilots of aerobatic, 3D or F3A-style fixed-wing models.

With conventional direct control, stick position primarily determines control-surface deflection. The resulting angular rate is then naturally dependent on airspeed, aerodynamic loading and the characteristics of the aircraft.

For example, the same aileron stick deflection normally produces a much faster roll at high airspeed than it does during a slow approach. For a traditional fixed-wing pilot, this is not necessarily an error that the flight controller should try to eliminate; it is part of the normal behavior of the aircraft.

What I would like to investigate is therefore something closer to:

pilot input -> direct surface command

plus

gyro -> corrective surface offset

rather than:

pilot input -> requested angular rate -> PID/FF -> surface command

The gyro correction could potentially be attenuated according to stick input so that it mainly suppresses unwanted disturbances while preserving the natural response of the aircraft during deliberate maneuvers.

The existing fixed-wing I-Term Lock / PID attenuation work moves ACRO somewhat in this direction by giving a more MANUAL-like feel during aggressive inputs, but I think it addresses only part of the underlying difference: ACRO remains fundamentally a rate-command controller.

This may also provide another way of approaching some of the yaw/turn-coupling problems discussed in this issue, since the controller would not necessarily interpret every naturally occurring yaw rate during a banked turn as an error relative to a commanded yaw rate.