iNavFlight/inav #5867: Support new version of NOXE F4 FC

FranzAr, 2020-06-21:
## Current Behavior
The NOXE F4 FC from Banggood has been updated to a new version labeled "V1". The "old" target NOX does not work on that new hardware version.

## Desired Behavior
Support new version "V1" as this FC is quite inexpensive and very small. Ideal for very limited space and light builds.

## Suggested Solution
Betaflight seems to support this FC with the target Flywoo F411 and on the Flywoo website there is an iNav version 2.3.0 downloadable for this target. I flashed this to the new NOXE board and it seems to work fine. Maybe this could be a good starting point.

## Who does this impact? Who is this for?
Everyone who needs a inexpensive and very small FC for iNav :-)

## Additional context
iNav download from Flywoo: https://flywoo.net/products/flywoo-goku-f411-micaro-stack
Noxe F4 at Banggood: https://www.banggood.com/de/20x20mm-Upgrade-Betaflight-F4-Noxe-V1-Flight-Controller-AIO-OSD-5V-8V-BEC-w-Barometer-and-Blackbox-for-RC-Drone-p-1310419.html

[image]

digitalentity, 2020-06-21:
Please have a look at https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md#new-target-additions

snaewe, 2020-06-22:
What, if someone would come up with a target config derived from betaflight for this FC? Any chance that would be merged ?

digitalentity, 2020-06-22:
As a baseline - community created targets may be merged, but are unlikely to be included in the release.

A one off exceptions are possible if a board is well designed, would be a good fit for INAV and there is a significant demand for support of the board.

moggiex, 2020-06-22:
Howdy @digitalentity ,

The board is a cheap 20x20mm board that even I picked because of the price, it has an OSD, 4 PWM channels and enough UARTs to be useful.

If I was to get one of these sent to you, would you be able to use it as a Dev board to make it compatible with iNav?

Matt

moggiex, 2020-06-22:
Actually @digitalentity ,

Reading the policy, if I went through list of common cheap boards in BG and sent you say up to 5 of them, would that be better overall?

Have checked the minimum specs and it would take a short while to list and cross reference the boards, then get you to double check them.

The boards like this one will get bought by people like me and the user above because of their price, specs and specifically for their size for light builds.

Is that something you would find helpful?

Matt

digitalentity, 2020-06-22:
Thanks for reaching out Matt, that would indeed be helpful. What boards do you have in mind?

moggiex, 2020-06-22:
Howdy,

The one above for sure and basically anything is £25 or less per board that isn't supported. 

The £25/$30 limit is a sensible one as the premium boards will be the likes of Matek and normally supported early. If you have any suggestions on specific ones let me know. I'll go do some searching now and see what I find.

Also are you able to send me an address and I'll get the NOXE board above sent to you asap too.

Matt

digitalentity, 2020-06-23:
@moggiex Thanks for the offer, please give me a ping at ksharlaimov@inavflight.com to discuss further

moggiex, 2020-06-30:
For anyone who is reading this and want at least 2.3.0 of iNav to run on this board, see here https://www.youtube.com/watch?v=lpzOU7lIsEg

@digitalentity , saw your reply earlier, have emailed you directly. 

Matt

tonyyng, 2020-06-30:
Can you also include SoftSerial support so we can use SmartAudio or Tramp components?

Thanks

snaewe, 2020-07-01:
The HEX file linked above does not work!
Based on the work by @SeanTheITGuy I did some changes and tests for the target. Look at my inav fork https://github.com/snaewe/inav/tree/add-FLYWOOF411
A small test is shown here: https://www.youtube.com/watch?v=pH0Ztvg2kkU

mooiweertje, 2020-07-01:
I cloned the add-FLYWOOF411 repo but wasn't able to find the FLYWOOF411 target.

snaewe, 2020-07-01:
@mooiweertje If you cloned my repo, you have to checkout the 'add-FLYWOOF411' branch first. Then 'make TARGET=FLYWOOF411'. HTH

moggiex, 2020-07-01:
Thanks gents. 

If you need an idiot to test with, just tag me a hex file :)

Matt

snaewe, 2020-07-01:
@moggiex https://github.com/snaewe/inav/releases/tag/0.1

mooiweertje, 2020-07-01:
Looking good. I will walk through your changes and comment here.
I did notice at first glance you have only one TIM_USE_FW_MOTOR in target.c but there should be 2. 
You can add your target to the target list in make/targets.mk.
Add a document docs/Board-FlyWooF411.md

@moggiex Probably snaewe tested a lot already but all IO needs testing. Flight functionality will work normal. Change is mainly about the CPU communicating through which pins to what peripherals. UART, ESC, servo, BARO etc etc. So all peripherals need testing, simply work or not work.

@digitalentity This change is looking good to my opinion and needs maybe minor ajustments and documentation. Is there a chance this change can go with 2.5.x in the near future?
@snaewe is there anything blocking your progress I could lend you a hand?

snaewe, 2020-07-01:
@mooiweertje @SeanTheITGuy is also working on this (I based my changes on his). We need to synchronize that.

Re. 2 motors: Really? Why do we want two motors with this tiny board? Will that still allow 3 servos + 1 motor ?

mooiweertje, 2020-07-01:
It's just how INAV works with two motors on FixedWing.

moggiex, 2020-07-01:
> It's just how INAV works with two motors on FixedWing. You can use only one of course but the mapping should have 2.

Wheres the head bang emjoi? 

PS. Yes I know it's written into the base code and that its timer related. But really? How many people actually use two motors, let alone diff thrust and use it in the air anyway? Huge waste a PWM channel, especially on these smaller boards with limited outputs. Its been a feature waiting for a cause for years. Meh, rant over ;)

mooiweertje, 2020-07-01:
P.S.
It's not the big thing you assume it to be now. I'm not a core iNav developer but just a developer that likes this board for it's obvious reasons, small, 2UARTS, BARO, cheap, etc. I'm just trying to conform to standards to quickly release without fuss. I have 2 of these boards I want to use soon. I didn't realize either the new Noxe v1 is so different than the old Noxe v0.
I don't know why 2 motors is default for FixedWing in the confiigurator mixer tab. Maybe you can create a separate RFC for the two motor default for FW in the mixer tab in configurator but if you make it part of this change it may slow down release.

mooiweertje, 2020-07-01:
This is the mapping of an AIRBOTF4. Markt the OR symbol '|' between the TIM_USE_** stuff.
S1 can do either MultiCopter motor or FixedWing motor. Quad or plane basicly. The same goed for S2. S3 and S4 Can be MultiCopter motor or FixedWing servo. This is the mapping that is the coding standard as it seems and complies with the defaults in the configurator mixer tab. When I configured S2 of an AIRBOTF4 to drive a servo  in the mixertab in configurator it did so without any issues.
    DEF_TIM(TIM3,  CH3, PB0,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,                                  0, 0), // S1_OUT D1_ST7
    DEF_TIM(TIM3,  CH4, PB1,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,                                  0, 0), // S2_OUT D1_ST2
    DEF_TIM(TIM2,  CH4, PA3,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,                                  0, 1), // S3_OUT D1_ST6
    DEF_TIM(TIM2,  CH3, PA2,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,                                  0, 0), // S4_OUT D1_ST1
    DEF_TIM(TIM5,  CH2, PA1,  TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO | TIM_USE_ANY, 0, 0), // S5_OUT / LED
    DEF_TIM(TIM1,  CH1, PA8,  TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO,               0, 0), // S6_OUT

snaewe, 2020-07-01:
I assume creating two target variants (1 motor vs. 2 motors) would be an option

mooiweertje, 2020-07-01:
No just one with the standards is fine. You can setup what even you want yourself in the configurator mixer tab.

snaewe, 2020-07-01:
> No just one with the standards is fine. You can setup what ever you want yourself in the configurator mixer tab.

Not if I want 3 servos and one motor and two of the four available outputs are "marked" as motor outputs, or am I missing something?

snaewe, 2020-07-01:
> > No just one with the standards is fine. You can setup what ever you want yourself in the configurator mixer tab.
> 
> Not if I want 3 servos and one motor and two of the four available outputs are "marked" as motor outputs, or am I missing something?

I compiled a version with two outputs marked as motor, setup the mixer (1 motor + 3 servos) and got "PWM...channel...blah" error in cli "status" -> No go.

mooiweertje, 2020-07-01:
I'm missing the details too but this is the standard and it works the way you want it to work. Just do S1 and S2 TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR and S3 and S4  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO. You will see you can configure what you want in the configurator mixer tab.

snaewe, 2020-07-01:
> I'm missing the details too but this is the standard and it works the way you want it to work. Just do S1 and S2 TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR and S3 and S4 TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO. You will see you can configure what you want in the configurator mixer tab.

Ehh...Nope. Doesn't work

mooiweertje, 2020-07-01:
I've compiled and installed a version with this mapping. It seems to work fine.
    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 1), // S1_OUT 2,1
    DEF_TIM(TIM1, CH2, PA9,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 1), // S2_OUT 2,2
    DEF_TIM(TIM1, CH3, PA10, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S3_OUT 2,6
    DEF_TIM(TIM3, CH3, PB0,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S4_OUT 1,7

snaewe, 2020-07-01:
This is my version:
    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 1), // S1_OUT 2,1
    DEF_TIM(TIM1, CH2, PA9,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 1), // S2_OUT 2,2
    DEF_TIM(TIM1, CH3, PA10, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S3_OUT 2,6
    DEF_TIM(TIM3, CH3, PB0,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S4_OUT 1,7

and it doesn't work!

[image]

moggiex, 2020-07-01:
Note: I have a board here and servos connected, can easily test if needed gents.

mooiweertje, 2020-07-01:
Ah yes, I have the PWM output init error: Not enough servo outputs/timers message too. And the servo on your board doesn't respond?

mooiweertje, 2020-07-01:
[inav_2.5.0_FLYWOOF411.hex.zip](https://github.com/iNavFlight/inav/files/4857029/inav_2.5.0_FLYWOOF411.hex.zip)

snaewe, 2020-07-01:
@moggiex https://bintray.com/snaewe/generic/download_file?file_path=inav_2.5.1_FLYWOOF411.hex

snaewe, 2020-07-01:
> Ah yes, I have the PWM output init error: Not enough servo outputs/timers message too. And the servo on your board doesn't respond?

I really didn't test...but what do you expect?

mooiweertje, 2020-07-01:
I expect it not to respond.

mooiweertje, 2020-07-01:
This doesn't give the error but stil one motor:
    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 1), // S1_OUT 2,1
    DEF_TIM(TIM1, CH2, PA9,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 1), // S2_OUT 2,2
    DEF_TIM(TIM1, CH3, PA10, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S3_OUT 2,6
    DEF_TIM(TIM3, CH3, PB0,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S4_OUT 1,7

mooiweertje, 2020-07-01:
This seems to work too and seems like a good mapping to me:
    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 1), // S1_OUT 2,1
    DEF_TIM(TIM1, CH2, PA9,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR | TIM_USE_FW_SERVO,  0, 1), // S2_OUT 2,2
    DEF_TIM(TIM1, CH3, PA10, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S3_OUT 2,6
    DEF_TIM(TIM3, CH3, PB0,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S4_OUT 1,7

mooiweertje, 2020-07-01:
Yes that should make all of us happy.

snaewe, 2020-07-01:
Did you test it? @moggiex and I are currently testing and even though we don't get that "status" error, there's no signal at the output!

SeanTheITGuy, 2020-07-01:
I don't think you can mix motors and servos on the same timer because they
want totally different frequencies.  Hence one motor and three servos in fw
mode.

mooiweertje, 2020-07-01:
My finding is the same about mixing servo's and motor mappings in target.c. I thought there was some kind of algorithm to choose the right mapping based on the mixer tab settings.
Nevertheless all the targets have S1 and S2 TIM_USE_FW_MOTOR......
I now think that the servo will turn on S2 since I have done that in the past with other boards, But not sure.

SeanTheITGuy, 2020-07-01:
You can't look at what other targets do, because other targets use the
timers differently.  Look at what timer you are using per output, and group
motors on timers separate from other items like PPM, UART and SERVO
outputs.  This is why some targets have motor outputs seemingly randomly
allocated among the various S[X] outputs.

tonyyng, 2020-07-01:
Do you think it would be possible to add a SoftSerial port?

Thanks

snaewe, 2020-07-01:
> Do you think it would be possible to add a SoftSerial port?

SoftSerial is enabled on TX2 but I haven't tested it yet.

SeanTheITGuy, 2020-07-01:
I will often repurpose the PPM pin for Softserial TX, to use as a
smartaudio port, leaving the UARTS intact for CRSF and GPS.  All that needs
to be done for that is to set define the SSTX pin in target.h as the PPM
pin, and SSRX as "NONE".

tonyyng, 2020-07-01:
> > Do you think it would be possible to add a SoftSerial port?
> 
> SoftSerial is enabled on TX2 but I haven't tested it yet.

Why would you put a Softserial port on a UART? The typical usage would be a receiver on TX1, GPS on TX2, and VTX control (Tramp, SA) on Softserial.  Could Softserial be put on something like an LED pad?

SeanTheITGuy, 2020-07-01:
This is typical with INAV.  It's done so you can have two RX pins, or two
TX pins instead of TX/RX.  With IBUS/SBUS + GPS, for example, you can put
both on one UART if you use softserial to repurpose both pins as an RX.

mooiweertje, 2020-07-01:
I need 2 UARTS for DJI FPV and GPS so that leaves PPM for a receiver.

SeanTheITGuy, 2020-07-01:
DEF_TIM(TIM9, CH1, PA2,   TIM_USE_PPM,   0, 0), // PPM IN

PA2 is PPM.

mooiweertje, 2020-07-01:
Just cosmetics but I think BUS_SPI3 should be renamed to BUS_SPI2 which sounds more logical to me and all seems to work fine when I tried.

SeanTheITGuy, 2020-07-01:
Sounds good. I just used the bus numbering from the Betaflight wiki on that board.  Since more work was done on the target that utilized mine as a base, that should probably be suggested for incorporation into the main project.

mooiweertje, 2020-07-01:
Maybe talk to digital entity to check if it's ok to just have one FW motor and 3 servo. It makes sense to me. It's for the drift I assume?

@SeanTheITGuy I know PPM is on PA2 but which connector or pad is PA2 connected to? This pin seems not to be connected to anything. If this is so I would like to use the RSSI pad for PPM. I don't use RSSI and I think all people nowadays use channel 16 or 8 foor RSSI.
And so far I have everything I want so I try not to go into the whole timer and channel circus because that takes a lot of time.

moggiex, 2020-07-01:
> DEF_TIM(TIM9, CH1, PA2,   TIM_USE_PPM,   0, 0), // PPM IN
> 
> PA2 is PPM.

I thought this board was SBUS. Or can it accept PPM instead? 

Excuse the daft question, you may just be able remap all the pins to whatever is desired. :)

mooiweertje, 2020-07-01:
@moggiex Daft questions are the ones not being asked I hear a lot. I agree. I would like PPM mapped to PB1 which is the RSSI pad now and remove RSSI since RSSI commonly is fed by the receiver channel 8 or 16. But making a lot of changes would confuse people cause the FC manual and writing on the actual board itself would be wrong.

mooiweertje, 2020-07-01:
The actual CPU is an STM32F411CEU6 which datasheet can be found here: https://www.st.com
/resource/en/datasheet/stm32f411ce.pdf which is a massive load of info for me. But it does contain some stuff I use sometimes. Out CPU is the 48 pin version Shown here: 
[image] Where you can find pin PA2 for instance. If you are handy you can solder the PPM wire from you receiver straight to this pin and it word work.

mooiweertje, 2020-07-01:
What is the USE_IMU_ICM20689 for? Betaflight has it too but I don't see the chip on the board and there is no external SPI bus.

mooiweertje, 2020-07-01:
I see some VTX switcher in Betaflight. Is that connected to the C-control pad? Is this smartaudio/tramp? On PB5?

#define USE_PINIO
#define PINIO1_PIN              PB5  // VTX  switcher
//#define PINIO2_PIN              PA15 // Camera switcher
#define USE_PINIOBOX

mooiweertje, 2020-07-01:
I've used my multimeter and I think I have continuity between PA2 and the CAM-C pad. Which would mean the CAM-C pad is PPM. Which is great.

mooiweertje, 2020-07-01:
I've run through all the target files for this board and I think all is well. Nothing critical to mention just the above suggestions. Thanks for sharing I will start using it in a few weeks or so. I am busy the next two days.

You can add your target to the target list in make/targets.mk to make it appear in the target message.
Add a document docs/Board-FlyWooF411.md

mooiweertje, 2020-07-01:
PA2 is the UART2_TX_PIN as well as PPM. So I guess what you configure in inav configurator is what the pin will do.

[348 more replies]

closed 2022-05-12 by DzikuVx