iNavFlight/inav #5868: Jeti duplex telemetry does not work ...

Turtletrumpet, 2020-06-21:
Hello,

i use a Matek F405 wing flight controller with a Jeti Duplex dc16. for the receiver i use the UART2 with RX2/TX2 and as protocol jetiexbus. The servo deflections seem to be transmitted correctly, but unfortunately the telemetry data transmission is missing.
I have a 2.2kOhm resistor between rx2 and tx2, but this did not improve anything.
If you need more information, please ask.
I hope you can help me.

regards,

Turtle

Turtletrumpet, 2020-06-21:
Ooops, I forgot to mention that I am using the current version 2.5.1 of iNav.

RoadyFPV, 2020-06-21:
thats true, i started a request for implementation of the optimazed code. #4711

Turtletrumpet, 2020-06-28:
Hello,
is there any new information in the meantime, when the telemetry for Jeti will work?
Thanks!

regards,

Dirk

RoadyFPV, 2020-06-29:
I don't know exactly. But it is being worked on.

Turtletrumpet, 2020-07-10:
Is there a new version to test available?

closed 2021-07-28 by DzikuVx