iNavFlight/inav #5873: Support UP-Flow LC302 Optical Flow Module

digitalentity, 2020-06-23:
An alternative to PMW3901 flow sensor. Available for purchase from AliExpress http://s.click.aliexpress.com/e/_d7BadZG

Sebbb, 2020-12-16:
Here is the manual with the protocol for this LC302 module (unfortuantely in chinese, but I was able to read it with Google Translate)

https://github.com/Sebbb/inav/raw/lc302/docs/1__________________(LC-302-3C)V1.1.pdf

The data structure   of the external output of the optical flow module is defined as follows:

```
typedef struct   optical_flow_data
{
int16_t flow_x_integral; // The cumulative   displacement of X pixels in the cumulative time (radians*10000)
                                     // [Divide by   10000 and multiply by height (mm) and then the actual displacement (mm)]
int16_t flow_y_integral; // The cumulative   displacement of Y pixel points in the cumulative time (radians*10000)
                                      // [Divide by   10000 and multiply by height (mm) and then the actual displacement (mm)]
uint16_t   integration_timespan; // The cumulative time from the last optical flow data   transmission to the current optical flow data transmission (us)
uint16_t   ground_distance; // Reserved. The default is 999 (0x03E7)
uint8_t valid; // Status value: 0   (0x00) means optical flow data is not available
                     //245(0xF5) is   available for optical flow data
uint8_t version; //version number

```

1 | 0xFE | The start   identifier of the packet (fixed value 0xFE)
-- | -- | --
2 | 0x0A | Number of bytes of   optical flow data structure (fixed value 0x0A)
3 | low byte of   flow_x_integral | The cumulative   displacement of X pixel points in the cumulative time,
  |   | (radians*10000)
  |   | [Divide by 10000   and multiply by the height (mm) to obtain the actual displacement (mm)]
4 | high byte of   flow_x_integral |  
5 | low byte of   flow_y_integral | The cumulative   displacement of Y pixel points in the cumulative time,
  |   | (radians*10000)
  |   | [Divide by 10000   and multiply by the height (mm) to obtain the actual displacement (mm)]
6 | high byte of   flow_y_integral |  
7 | low byte of   integration_timespan | The optical flow   data sent last time to the optical flow data sent this time
  |   | Cumulative time   (us)
8 | high byte of   integration_timespan |  
9 | low byte of   ground_distance | Reserved. The   default is 999 (0x03E7)
10 | high byte of   ground_distance |  
11 | valid | Status value: 0   (0x00) means optical flow data is not available,
  |   | 245 (0xF5) is   available for optical flow data
12 | version | Version number of   the optical flow module
13 | Check value Xor | Optical flow data   structure (Byte 3~Byte 12) 10 bytes
  |   | XOR
14 | Packet tail 0x55 | The end   identifier of the data packet (fixed value 0x55)

Sebbb, 2021-01-07:
I have this module, so if you need somebody to test, I can do that.

oldmanbluntz, 2026-10-23:
I assume that support for this has still not been added?