iNavFlight/inav #5875: Rounding of way point turns for fixed wing

moggiex, 2020-06-23:
## Current Behavior

When using Waypoints in a FW model, when it reaches a WP, it banks like a processed squirrel on crack after his next nut if the angle to the next WP is too great ( > 90 degrees) :D 

In 2.5.1 **nav_fw_control_smoothness**  was added to smooth things out for FW nav modes. This works as expected. YAY!

However it has an undesired effect (assumed) as it works for all NAV modes, so in 3D Cruise a model that was "tight" and reactive to fly (in 2.4), now just wanders around the sky if this value is too high (eg 7 or more) as it doesn't correct fast enough especially in a strong wind.

For absolute clarity I was looking for this to smooth out the harsh turns when in WayPoint mode. Which it does, but with the effect above. So now I know I was wrong, **nav_fw_control_smoothness** is a global setting, all good, that works as expected.

## Desired Behavior

My feature request is that can we have some "rounding" on the turns for FW in waypoint mode so the turns are not so harsh?

You can see how harsh the turns are in this video near the end https://www.youtube.com/watch?v=Y1AZMsZJCXo 

## Suggested Solution

Adding a setting is something like **a value that allows for a turn radius**, say 30 metres to make the turn at a WP a lot more smooth for FW models, rather than banking like a processed squirrel on crack after his next nut if the turn is too tight to the next WP :D 

## Who does this impact? Who is this for?

Those using waypoints with a FW model.

## Additional context

Commits here https://github.com/iNavFlight/inav/pull/5655

moggiex, 2020-06-23:
This new setting is different to **nav_wp_radius** which is defined as 

> Waypoint radius [cm]. Waypoint would be considered reached if machine is within this radius

This is the turn radius when the FW model reached the way point (whose size is defined by **nav_wp_radius**).

Matt

breadoven, 2022-11-27:
Considered closed by https://github.com/iNavFlight/inav/pull/8234.

closed 2022-11-27 by breadoven