iNavFlight/inav #5877: TBS Crossfire Telemetry Sensors Problem

cliffkhew, 2020-06-23:
I had an issue with the TBS Crossfire Micro TX & Nano RX with this version.Currently using Omnibus F4 PRO V2,CRSF TX & RX to UART#1.The radio discovered sensors many are missing & some are not displaying the correct values.The previous version like 2.4.0 & older versions don't have this issue,works perfectly.

Jetrell, 2020-06-26:
@cliffkhew I loaded 2.5.1 onto an old Omnibus F4Pro V3, that has a TBS Nano RX. 
OpenTX showed that all my senors are there and they also appeared in iNav Telemetry. 
Have you tried to re-run `Discover sensors`?

cliffkhew, 2020-06-26:
Yes,of course need to delete all sensors and discover the sensors again for new setup.But the sensors I got is not completed.Can you confirm that the sensors are all found?The completed should have 23 sensors-RxBt,Curr,Capa,Bat_,1RSS,2RSS,RQly,RSNR,RFMD,TPWR,TRSS,TQly,TSNR,FM,GPS,GSPD,Hdg,Alt,Sats,VSpd,Ptch Roll,Yaw.

Jetrell, 2020-06-26:
Yes. I can confirm that all the ones you have listed are being received. 21 in total.
I also forgot to mention that I was using UART 1 as well.

cliffkhew, 2020-06-26:
Im using Omnibus F4 PRO V2,i think V3 almost identical.It is great that u have no issue.I will try to upload 2.5.0 again to see it is working or not...btw my V3 fc just arrived will try it too.
Thks for reply,have a good flight:)

cliffkhew, 2020-07-01:
All right,I reflashed the latest firmware v2.5.1 to my Omnibus PRO V2 board,made new configuration without diff all config.Delete all discovered sensors & discovered new sensors.23 sensors appereared completely without any issue.Thank Jetrell replied to me and encouraged to solve the issue.

gcl2000, 2020-12-07:
Hi. How do you resolve the issue of the telemetry not showing either VBAt or RXBt in discovered sensors in tbs micro? I am in latest BF 4.2. Any help please.

cliffkhew, 2020-12-07:
For BF,I didn't try for it.I have iNav firmware v2.5.1 that without any issue.Probably the BF firmware issue.

closed 2020-12-11 by DzikuVx