iNavFlight/inav #5888: Autotune Mode for inav

Seeelefant, 2020-06-27:
## Current Behavior

Currently PID etc. tuning is a huge manual effort.

## Desired Behavior
Implement AUTOTUNE mode like in Arducopter: https://ardupilot.org/copter/docs/autotune.html

## Suggested Solution

"AutoTune attempts to automatically tune the Stabilize P, Rate P and D, and maximum rotational accelerations to provide the highest response without significant overshoot. Copter needs to be “basically” flyable in AltHold mode before attempting to use AutoTune as the feature needs to be able to “twitch” the copter in the roll and pitch axis."
## Who does this impact? Who is this for?

## Additional context

Black6spdZ, 2020-07-18:
didn't it used to be in INAV?

Seeelefant, 2020-08-04:
As far as I know only for planes.

closed 2020-12-11 by DzikuVx