iNavFlight/inav #5890: Incredible skyline drift

dsaboot, 2020-06-28:
## Current Behavior
Today it was really annoying, the horizon angle went 35 degrees or more, the plane simply could not fly in Angle mode, I switch to manuall mode and land, but if it happened far from takeoff and in the return home mode, who knows, it would back or not.
[https://youtu.be/xOhsyCsKXm0](url)
it's already set imu_acc_ignore_rate = 10
[image]

Inav 2.5.1
Matek F411-WING
Wing Z84
PID and filters setting from Z-84 profile (aplyed Z84 profile in configurator)

sdfsdf107, 2020-07-01:
It seems that this is a know bug.
git checkout remotes/origin/dzikuvx-bno055-secondary-imu

dsaboot, 2020-07-01:
It's  a know bug - then horizon drift near 10deg, but I have 35-45deg.

dsaboot, 2020-07-05:
After many tests, with very low freq filter setting, looks ok. Horizon drift not so high, as in past.

closed 2020-07-05 by dsaboot