iNavFlight/inav #5891: Pre-Arming Checks. 'Settings not validated' ### Error:Invalid setting: dterm_lpf2_hz'

jclaver, 2020-06-28:
Matek F765 v2.5.1 configured for flying wing and using SBUS from DJI FPV Unit and DJI Tx
Inav Configurator Version 2.5.0.RC2

CLI Status Report as follows (see ###ERROR...):

status
System Uptime: 455 seconds
Current Time: 2020-06-28T15:54:44.803+00:00
Voltage: 14.92V (4S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280, MAG=QMC5883, PITOT=ADC
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Manufacturer 0x2, 30748672kB, 01/2014, v1.2, 'SA32G'
Filesystem: Ready
Stack size: 6144, Stack address: 0x20020000, Heap available: 1820
I2C Errors: 45, config size: 5923, max available config: 32768
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = ADC 4
System load: 3, cycle time: 1002, PID rate: 998, RX rate: 101, System rate: 9
Arming disabled flags: ARMSW CLI SETTINGFAIL
### ERROR: Invalid setting: dterm_lpf2_hz
VTX: not detected
#

stronnag, 2020-06-28:
See wiki for solution:
https://github.com/iNavFlight/inav/wiki/%22Something%22-is-disabled----Reasons

This is usually caused by pasting obsolete settings.

jclaver, 2020-06-28:
Hi - I haven't pasted anything...

jclaver, 2020-06-28:
Also where is the clue as to the problem?...
"Arming disabled flags: ARMSW CLI SETTINGFAIL"
or
"ERROR: Invalid setting: dterm_lpf2_hz"
or both???

DzikuVx, 2020-06-29:
`dterm_lpf2_hz` has invalid value. `get dterm_lpf2_hz` and then `set dterm_lpf2_hz = xxx` to set the value you want. Of course replace `xxx` with proper value

closed 2020-06-29 by DzikuVx

jclaver, 2020-06-29:
The value I want?  Sorry but I'm not a developer... I just built a Matek board and was 'mistakenly' hoping I could commission it with iNav without too much hassle but instead I'm supposed to know what 'dterm_lpf2_hz' means?!? (I searched for it but it does not appear anywhere).  One further comment as an iNav user (but one familiar with Arduplane and Betaflight too) and someone that has commissioned other flight controllers...  I find it pretty weird every time I lodge an issue such as this... the terse responses I get as if I'm some newbie idiot...  This after I SPENT 2 HOURS + SORTING OUT USB SERIAL DRIVER contentions.  Stuff we shouldn't have to do as end users... and which are far more impacting than 'dterm_lpf2_hz'.