iNavFlight/inav #5893: Quadcopter on iNav 2.1.0: flyaway during RTH

DiscoMan18, 2020-06-28:
Today I made a couple of successful takeoffs and landings without reconnecting the battery. During all the flight I had more than enough of sats, 19-22, home point was successfully recorded, compass calibrated etc. There were no issues at all. 

During the last flight I lost RC Link only 60 meters away due to problems with my RC transmitter, got a message on OSD: FS! RTH, En Route to Home, then the RTH procedure has started. The quadcopter gained 60 meters of altitude that were set in the settings as RTH Altitude, but didn't start making its way to home. It has been just flying away down the wind, while I have been trying to regain the control. At 300 meters of distance the Emergency Landing procedure has started, in full accordance with the setting of RTH Abort Threshold in CLI, but at this moment I managed to regain the control, flew back and landed the quad. 

Please help me to understand what's happened and why RTH didn't function as expected.  

Link to my settings: https://pastebin.com/p6Z8antc 

- FC Board name and vendor:
- INAV version string:

# version
# INAV/MATEKF405 2.1.0 Feb 25 2019 / 16:58:21 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
[Flyaway log.TXT.zip](https://github.com/iNavFlight/inav/files/4842981/Flyaway.log.TXT.zip)

DiscoMan18, 2020-06-29:
Anyway please check for a possible bug with RTH mode. It's not working correctly while Home Reset mode is active. Yesterday I got a failsafe while being only 70 meters from takeoff point. The quad ascended to preset RTH altitude, 60 meters in my case, but then started to drift down the wind, flying away from me. On the OSD correct distance and direction to the takeoff point were shown. 

Before failsafe fired, there were two flight modes active: POSHOLD and HOME RESET.

stronnag, 2020-06-29:
Please reproduce the problem on a contemporary version of inav, 2.1.0 is no longer supported.

closed 2020-06-29 by digitalentity