iNavFlight/inav #5894: Throttle is force limited below 12.5vDC

HerpDerpula, 2020-06-29:
Current Behavior
I noticed the current being drawn from LiIon cells seems to be intentionally limited when approaching 12vDC. It appears that iNav is doing this intentionally. 

Battery is configured to be 4S in configuration.

Steps to Reproduce
1. Fly to a safe altitude where you can glide home
2. Fly until batteries start dipping below 13vDC (4S)
3.Around 12.5vDC (may vary) throttle will begin to cut, and the plane will be unable to maintain altitude
4. Switch out of automated modes and land at home

Alternate steps:
1. Plug in a battery below 13vDC (4S)
2. Increase throttle until throttle begins to cut

## Expected behavior
The user in the loop should have the capability to override automation

## Suggested solution(s)
Allow configuration changes to this behavior

- FC Board name and vendor: Matek F405-Wing
- INAV version string:  INAV/MATEKF405SE 2.4.0 Feb 11 2020 / 22:58:10 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

DzikuVx, 2020-06-29:
INAV does not have any throttle limiting functions enabled by default.  You might have enabled something but it was your action. Please provide a configuration dump

breadoven, 2020-06-29:
Isn't this the ESC limiting output to protect the battery ?

HerpDerpula, 2020-06-29:
It very well may be the ESC, I didn't think about that. That's annoying if so. I'll need to determine the ESC model number and find a way to passthrough throttle on bootup. 
`dump` all

# version
# INAV/MATEKF405SE 2.4.0 Feb 11 2020 / 22:58:10 (bcd4caef9)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 950 2050 1508 -110
servo 4 950 2050 1498 110
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# logic
logic 0 0 0 0 0 0 0 0
logic 1 0 0 0 0 0 0 0
logic 2 0 0 0 0 0 0 0
logic 3 0 0 0 0 0 0 0
logic 4 0 0 0 0 0 0 0
logic 5 0 0 0 0 0 0 0
logic 6 0 0 0 0 0 0 0
logic 7 0 0 0 0 0 0 0
logic 8 0 0 0 0 0 0 0
logic 9 0 0 0 0 0 0 0
logic 10 0 0 0 0 0 0 0
logic 11 0 0 0 0 0 0 0
logic 12 0 0 0 0 0 0 0
logic 13 0 0 0 0 0 0 0
logic 14 0 0 0 0 0 0 0
logic 15 0 0 0 0 0 0 0

# gf
gf 0 0 -1 0 0 0 0
gf 1 0 -1 0 0 0 0
gf 2 0 -1 0 0 0 0
gf 3 0 -1 0 0 0 0
gf 4 0 -1 0 0 0 0
gf 5 0 -1 0 0 0 0
[... 1364 lines cut ...]
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# battery_profile
battery_profile 3

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

# `

DzikuVx, 2020-06-30:
Since you do not have `THR_VBAT_COMP` enabled, INAV is not doing any voltage-throttle adjustments. Maybe ESC then

closed 2020-06-30 by DzikuVx