iNavFlight/inav #5895: F722S resources wrong?

MaxMihawk, 2020-06-29:
# Current Behavior

resource shows very much FREE adresses
## Steps to Reproduce

1. flash MAMBA F722 firmware
2. click "3-7 inch copter" if INAV asks
3. go to CLI
4. type "resource"

## Expected behavior

see below

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: Mamba F722S
- INAV configurator version: 2.5.0
- INAV target version: 2.5.1 MAMBA F722 & 2.4.0 MAMBA F722

This is how it looks like in BF:
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Jun 14 2020 / 03:05:26 (8f2d21460) MSP API: 1.43

# config: YES
# board: manufacturer_id: DIAT, board_name: MAMBAF722

# resource
resource BEEPER 1 B02
resource MOTOR 1 C08
resource MOTOR 2 C09
resource MOTOR 3 A08
resource MOTOR 4 A09
resource PPM 1 B09
resource LED_STRIP 1 B03
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource LED 1 C15
resource LED 2 C14
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource CAMERA_CONTROL 1 B08
resource ADC_BATT 1 C01
resource ADC_RSSI 1 C02
resource ADC_CURR 1 C03
resource PINIO 1 B00
resource FLASH_CS 1 A15
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04

```

This is how it looks in INAV:
```
# version
# INAV/MAMBAF722 2.5.1 Jun 16 2020 / 13:07:33 (098a574b3)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# resource
IO:
----------------------
A00: SERIAL4 UART TX
A01: SERIAL4 UART RX
A02: FREE 
A03: FREE 
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: FREE 
A09: FREE 
A10: FREE 
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: FLASH CS
B00: FREE 
B01: FREE 
B02: BEEPER OUT
B03: FREE 
B04: FREE 
B05: SPI3 MOSI
B06: SERIAL1 UART TX
B07: SERIAL1 UART RX
B08: FREE 
B09: FREE 
B10: I2C2 SCL
B11: I2C2 SDA
B12: OSD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: FREE 
C03: FREE 
C04: MPU EXTI
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: SPI3 SCK
C11: SPI3 MISO
C12: FREE 
C13: FREE 
C14: LED2 OUT
C15: LED1 OUT
D02: FREE 
```

I´m missing so much UARTS, Motor resources etc. Something wrong here... ?

MaxMihawk, 2020-06-30:
The guys appear magical if I use them. What a feature -.-" . Without the BF map I´ll never know, where is my PPM to remap SOFTSERIAL on it, for example. I´m not sure if this is rly a good idea.

```
# resource
IO:
----------------------
A00: SERIAL4 UART TX
A01: SERIAL4 UART RX
A02: SERIAL2 UART TX/RX
A03: FREE 
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MOTOR3 OUT
A09: MOTOR4 OUT
A10: FREE 
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: FLASH CS
B00: FREE 
B01: FREE 
B02: BEEPER OUT
B03: LED STRIP OUT
B04: FREE 
B05: SPI3 MOSI
B06: FREE 
B07: FREE 
B08: FREE 
B09: FREE 
B10: I2C2 SCL
B11: I2C2 SDA
B12: OSD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: FREE 
C03: ADC CH3
C04: MPU EXTI
C05: FREE 
C06: FREE 
C07: SERIAL6 UART RX
C08: MOTOR1 OUT
C09: MOTOR2 OUT
C10: SPI3 SCK
C11: SPI3 MISO
C12: SERIAL5 UART TX
C13: FREE 
C14: LED2 OUT
C15: LED1 OUT
D02: SERIAL5 UART RX
```

digitalentity, 2020-07-01:
This is by design. If an output is not used, it's not mapped to CPU pins. INAV doesn't have remapping (and is unlikely to have). The `resources` command provides an overview of **currently used** hardware resources and exists only for debugging purposes.

closed 2020-07-01 by digitalentity