iNavFlight/inav #5896: Matek F411 WSE Tramp VTX Control not working on Softserial 1

SnappyFPV, 2020-06-29:
INav Version tested on 2.3, 2.4, 2.5.1 
SoftSerial 1 used for VTX Control
VTX tested: Holybro Atlatl HV v1.2, Eachine Nano VTX
Both working on regular UART TX but not on SoftSerial 1
CLI Setting:
serial 30 4096 115200 115200 0 115200

Tested with a TBS Unify Pro32 and SmartAudio on SoftSerial 1 - working without issues.

SnappyFPV, 2020-06-30:
Did also test several baudrates with the Port. 19 - 115 - no success

mf0o, 2020-10-06:
I used the Eachine Nano VTX at the ST1 of F411-WSE (v2.5.2) with success. Could change params via OSD and used to change output power inflight through LC+GF.
Also with: 
serial 30 4096 115200 115200 0 115200

paulpenney, 2020-12-11:
With the GEPRC Baby Croc Analog F411 AIO and STABLE 600mw VTX, I can set Power level 1-4, but not 5.

Vereshchaginag, 2021-03-11:
Same issue, but found some strange things happening:
After powering up from lipo Tramp not working, but if I unplug vtx connector from tramp and conncting it back some magic happens and tramp becomes visible and can be configured via FC...

closed 2021-11-14 by DzikuVx

Isopodus, 2022-01-28:
Any updates on this? I can still see this not working on my F411-WSE and Diatone Mamba Ultra Tiny 400mW. Both SF1 and SF2 are not working.

cmanley, 2022-09-01:
Why was this bug closed?