iNavFlight/inav #5899: MATEK405STD no stick controls after arming

MeesterSee, 2020-06-30:
Running a matekf405std flashed with inav 2.5.1 . Setup for 10inch quad preset.
All calibrations done. 
Once armed props spin. Throttle can be raised to 100% but there are no control movements in bwtween.

Then reflashed down to inav 2.4 and exact same setup performed.
Quad is armed and props spin and throttle raised. It has full control movements and is able to be hovered and punched out.
Good luck guys. You are all legends!!
Thank you

digitalentity, 2020-06-30:
You mean it hovers (doesn't flip over), but no stick control. Can you please post a dump?

MeesterSee, 2020-06-30:
There is no hover in inav 2.5.1. If throttle is increased quad gently flips
over on ground. Does not lift off ground as if no self levelling activated.

I have no dump from 2.5.1 due to flashing back to 2.4. Sorry

digitalentity, 2020-07-01:
Without the non-working dump I'm afraid we will be unable to debug this. Try flashing 2.5.1 with full erase and configure without applying any presets.

MeesterSee, 2020-07-01:
Thats fine Konstantin.
I appreciate you getting back to me.
I will keep flying on 2.4 for now. Whenever a new version comes out i will
reflash.
I just wanted to let you  guys know of this issue if it comes up again.
Cheers

digitalentity, 2020-07-01:
Cool, thanks. I'm closing this for the time being then. If you stumble upon the same issue again - please reopen

closed 2020-07-01 by digitalentity