iNavFlight/inav #5900: Mag detected and correct but showing red

KushBallK, 2020-07-01:
Hi all, after a short flight of my drone, I tried to re-arm it to fly again and it wouldn't due to it not passing the hardware health check and a red mag indicator on INAV. However, the mag was still being detected and reading accurate values. Reconfiguring the mag and recalibrating didn't help either. Plugging in another identical gps/mag module let it pass the check, but its readings were very laggy. I've attached the log from the first flight. This was a new Matek m8q-5833 on a new Lumenier F7 Lux FC running INAV 2.5.0. Any ideas?
[LOG00011.TXT](https://github.com/iNavFlight/inav/files/4854975/LOG00011.TXT)

sdfsdf107, 2020-07-01:
Yes, I have issue on m8q-5833 and f722se, the mag is red in inav configurator. I also test with hglrc gps both on 5v and 4.5v. the 5883 also failed on iic. But I use another f722se , and use 4.5v and scl sda , connect to bno055. The iic works.

digitalentity, 2020-07-01:
Hi! Please provide complete `dump` and `version` and photos of how your mag is connected.

closed 2020-07-04 by no-response[bot]