iNavFlight/inav #5901: Can't free resources

carlybaby69, 2020-07-01:
I am unable to free resources in iNAV. For example I have the following resource:

B05: LED1 OUT

If I type:

resource LED1 OUT NONE

Then it simply lists out all the resources again as if I had simply typed resource. Expected is that I would see resource is freed message. Target is OmnibusF4V3Pro.

stronnag, 2020-07-01:
The `resource` command in inav only lists resources and is well documented. This is a feature.

DzikuVx, 2020-07-01:
There is no resource mapping in INAV

closed 2020-07-01 by DzikuVx

tfoutfou, 2020-07-04:
@carlybaby69 
you have to make a custom target and compile it

If you give me all the needed parameters i can do it for you if you dont know how

carlybaby69, 2020-07-04:
Hi,
Thanks! I’ve been making my own custom targets anyway as this board did not have an internal baro. 
I have no more UARTS and want to use Smart Audio therefore want to enable soft serial on the LED instead. If I go to resources I see this:
B05: LED1
I would therefore like to assign this as my soft serial port. The thing I don't understand is when I look at target.h I see the LED defined as:
#define LED0                    PB5
Do what should I be defining soft serial 1 as? The target.h file doesn't seem to map to what I see in resources. Any help appreciated. Thanks!------ 
Original Message ------

tfoutfou, 2020-07-04:
it s kind of a mess in the target.h file because there is many version of the board.  and i am not familiar with omnibus board.

LED PB5  is the onbard led indicating the status of the board 
You might want to use the LED STRIP pad right ? wich is PB6

but i would suggest you to use PWM5 pad or PWM6 pad  wich are respectively  PA1 and PA8

in target REMOVE this whole section:

> #if defined(OMNIBUSF4V3) && !(defined(OMNIBUSF4V3_S6_SS) || defined(OMNIBUSF4V3_S5S6_SS) || defined(OMNIBUSF4V3_S5_S6_2SS))
> #define USE_SOFTSERIAL1
> #define SOFTSERIAL_1_RX_PIN     PC6     // shared with UART6 TX
> #define SOFTSERIAL_1_TX_PIN     PC6     // shared with UART6 TX
> 
> #define SERIAL_PORT_COUNT       5       // VCP, USART1, USART3, USART6, SOFTSERIAL1
> 
> #elif defined(OMNIBUSF4V3_S6_SS)        // one softserial on S6
> #define USE_SOFTSERIAL1
> #define SOFTSERIAL_1_RX_PIN     PA8     // S6 output
> #define SOFTSERIAL_1_TX_PIN     PA8     // S6 output
> 
> #define SERIAL_PORT_COUNT       5       // VCP, USART1, USART3, USART6, SOFTSERIAL1
> 
> #elif defined(OMNIBUSF4V3_S5S6_SS)      // one softserial on S5/RX S6/TX
> #define USE_SOFTSERIAL1
> #define SOFTSERIAL_1_RX_PIN     PA1     // S5 output
> #define SOFTSERIAL_1_TX_PIN     PA8     // S6 output
> 
> #define SERIAL_PORT_COUNT       5       // VCP, USART1, USART3, USART6, SOFTSERIAL1
> 
> #elif defined(OMNIBUSF4V3_S5_S6_2SS)    // two softserials, one on S5 and one on S6
> #define USE_SOFTSERIAL1
> #define SOFTSERIAL_1_RX_PIN     PA1     // S5 output
> #define SOFTSERIAL_1_TX_PIN     PA1     // S5 output
> 
> #define USE_SOFTSERIAL2
> #define SOFTSERIAL_2_RX_PIN     PA8     // S6 output
> #define SOFTSERIAL_2_TX_PIN     PA8     // S6 output
> 
> #define SERIAL_PORT_COUNT       6       // VCP, USART1, USART3, USART6, SOFTSERIAL1, SOFTSERIAL2
> 
> #else                                   // One softserial on versions other than OMNIBUSF4V3
> #define USE_SOFTSERIAL1
> #define SOFTSERIAL_1_RX_PIN     PC8     // pad labelled CH5 on OMNIBUSF4PRO
> #define SOFTSERIAL_1_TX_PIN     PC9     // pad labelled CH6 on OMNIBUSF4PRO
> 
> #define SERIAL_PORT_COUNT       5       // VCP, USART1, USART3, USART6, SOFTSERIAL1
> #endif
> 

and then write :

> #define USE_SOFTSERIAL1
> #define SOFTSERIAL_1_RX_PIN     PA1
> #define SOFTSERIAL_1_TX_PIN     PA1
> 
> #define SERIAL_PORT_COUNT       5

it might do the trick
(in the configuration tab dont forget to enable CPU based Serial Port  , and then configure SmartAudio in the Port Tab for SoftSerial1)

tfoutfou, 2020-07-05:
and to clarify about PB5 vs B05  they are the same

just add a "P" , remove the "0"  and B05 become PB5.

Dont quote me on that but PB5 is the real name for the CPU pin , and B05 is to be presented to the human eye as a more readable format.

because if you have PB1 PB11 PB14 PB8
if you dont put the "0"  the order is  B1 B11 B14 B8
if you put the "0" the order is B01 B08 B11 B14  (classified in numerical order)