iNavFlight/inav #5903: Minor OLED issue

fiechr, 2020-07-01:
I've got some of these cheap OLED displays from AliExpress and it worked out of the box! :-)
I read the Wiki page and know that OLED support is only basic at the moment.

I'm using a Matek F405-WING with latest iNav Release 2.5.1

There seems to be some "bar" or graphics glitch on the bottom of the screen. I hope It's not a feature I don't understand... Besides that, the display seems to work fine.

[image]

Any chances or ideas of fixing this? Happy to provide more information, although apart from the pictures and the sellers information, my knowledge about the display module itself is really limited:
https://www.aliexpress.com/item/32896971385.html

I've tested the displays before using them on my wing, with an Arduino and the whole screen was "accessible" (no bars).

[image]
[image]

brainbubblersbest, 2020-07-07:
I remember that his Was not there in earlier releases. I noticed this on a newer build. But I thought my oled is broken.
Doesnt swapped the oled to multiple screen in earlier releases Till arming? Or was that in betaflight?

fiechr, 2020-07-09:
No, for me it only shows this screen shown in the image. But I remember reading something about what you describe (multiple screens), no idea... maybe that has been changed.
But, while armed, the display should change, but I did not test this yet since I'm currently replacing the motor and haven't armed it yet since installing the OLED.

KissmanCZ, 2020-07-28:
Same issue in my wing setup with F405 Wing... I am using dual color OLED and I have same "bar" at the bottom of the display.

ToniCupra, 2020-11-14:
is there a solution for the problem?

I found out with a old oled (bought this 2017) no Problem.

Only with the new one (2020) i 've got this Glitch

fiechr, 2020-11-14:
> is there a solution for the problem?

Unfortunately, no, AFAIK.

But I think it's interesting to know that it is related to newer hardware, since your older OLED works fine.

closed 2021-11-14 by DzikuVx

sensei-hacker, 2025-03-17:
These displays have 8 lines, each 8 pixels high.  It looks like 7 lines are written and the last line has not been written, so it's random pixels.

sensei-hacker, 2025-05-31:
I've learned there are a few different chips used for OLED drivers.

The SSH1106 needs more data sent to it than the SSD1306.  The SSD1306 just gets bytes for the pixels.  The SSH1106  has memory for four extra lines that are not displayed. I may do a PR to try to detect which kind of display is attached and adjust appropropriatrly.