iNavFlight/inav #5904: Home position arrow wrong

JCFPV, 2020-07-03:
INAV 2.5.1
MATEK 722PX
BN 880 GPS 
Selected Northern hemisphere for ground assistance 
I  have set up everything correctly, calibrated compass, acc set up modes  and still the home direction arrow is wrong it points down the road and as I turn it turns but in the wrong direction I have tried this with 2 different Flight controllers and same results. If I flash it with betaflight 4.2 it works great and is very accurate. But the whole point of getting a new flight controller was to use inav I like the way inav flies just need to get a solution to why the home direction is wrong also gps rth goes in the wrong direction every time I activate it and not always in the same direction it's like it doesn't know where it is

Mateyhv, 2020-07-03:
"...and still the home direction arrow is wrong it points down the road and as I turn it turns but in the wrong direction..."
Can you explain with a practical example what you mean? Thanks

JCFPV, 2020-07-04:
Well i am still having trouble getting inav to work correctly I pulled up mission control and got the compass orientation right and now when rth is enabled it flys back backwards at a angle and pitches so hard I am afraid its gonna fall out the sky also apparently it when into failsafe mode right when i armed it and tried to auto land except is was on the ground and when i tried to disarm it, it completely locked me out which in turn the quad still tried to land and shot up into the air and almost hit me right in the face. Why would it circle around and come back backwards but work great on BF has never has it locked me out.

Mateyhv, 2020-07-06:
There is an option in configurator to RTH without specificaly facing you, quads can return facing any direction. The max pitch and roll angle in angle mode (and NAV) are user set in the configurator too. Regarding failsafe on arming there is something definitely wrong in your radio link or setup so check it out even before mounting the props.

JCFPV, 2020-07-06:
Thanks I somewhat got it working it returns a lot better now I just kept recalibrating the compass and finally it is closer as far as the option in inav to select which direction it returns home do you by chance have the cli for that i wish i could just select which side is forward it seems weird when we have to set up so many settings to get it to fly right but we cant choose a side as the front so the quad knows which direction is forward if we are setting up 4 motors in a specific order why not choose the two motors that are forward eliminates the guess work instead of algorithms I thought as long as my fc was installed forward that would be enough I may be wrong considering I am a noob. I believe there was a lost rc link causing the failsafe I just wish I could override the failsafe when it locked me out I wasn't sure what to do or if I was getting it back but the drone did what it was supposed to it landed it's not its fault I was standing over top it ha ha lessons learned

closed 2020-07-06 by JCFPV

chucky-whiskers, 2022-09-03:
I agree that we should be able to automatically calibrate our magnetometer orientation the same way we calibrate accelerometers in iNAV. Then, calibrating the magnetometer orientation would not be a matter of guesswork. Calibrating the magnetometer using the 360° method seems to work fine, but my RTH tried really hard to fly away from home today. Luckily, I didn't try to failsafe and just tried momentary RTH switch. It is frustrating when the factory gives you the wrong orientation for the program.