iNavFlight/inav #5907: Matek405 wing all servos stucks after sticks movements

AndrewMishchenko, 2020-07-03:
Hello Guys. I'm new to all of this, and maybe I do something wrong! But IMHO it is really strange behaviour as it can crash my plane!
1. Turned on everything
2. Doing random sticks movements
3. All servos stuck and I can't do anything with them!
Really hard to explain but let me try to show you the video below!
https://drive.google.com/file/d/1rrQ4n5e8LQcUe5HP1OMvBl4S4aghJax3/view?usp=sharing

Inav 2.41/2.50
Firmware 2.51/2.50/2.41 Matek405SE

Thank you in advance!

digitalentity, 2020-07-06:
You're likely entering CMS. What is shown in the OSD once servos stop moving?

AndrewMishchenko, 2020-07-06:
@digitalentity I don't have it. And I don't know how can I check it in the configurator.

AndrewMishchenko, 2020-07-06:
And what is CMS inside Inav? Maybe I can turn it off somehow to check?

digitalentity, 2020-07-07:
CMS is a set of built-in OSD menus to adjust the common settings. Disable the OSD - that will also disable CMS

AndrewMishchenko, 2020-07-07:
@digitalentity I've disable the OSD and still same behaviour.

AndrewMishchenko, 2020-07-07:
Maybe it is a kind of a bug? Because Ialso can't arm. Everything is green, 8 satelites, GPS fix type: 3D. Added channel 6 for ARM in modes. But it still doesn't arm. Maybe related?

AndrewMishchenko, 2020-07-07:
@digitalentity finally I found and you are correct!
# satatus
Arming disabled flags: THR CLI CMS

So even CMS is disabled it enables is! Also I found it changes profiles! And it is strange because I've disabled it.

AndrewMishchenko, 2020-07-07:
Also figured out why it didn't arm! In Spektrum throttle minimum is 1160. And default value for **min_check** equals to 1100! So because of this it didn't arm!
**For others:**
set min_check = 1161
save

stronnag, 2020-07-07:
Better to read and apply the documentation, in particular https://github.com/iNavFlight/inav/blob/master/docs/Rx.md#receiver-channel-range-configuration

closed 2020-09-20 by stale[bot]