iNavFlight/inav #5910: Kakute F7 AIO / inav 2.51 - motors stop after ~1s in DSHOT

Seeelefant, 2020-07-05:
## Current Behavior
I'm trying new 2.51 build with Kakute F7 AIO

All works well when using Oneshot125/42, but when switching to DSOT 150/300/600, when arming motors start, but ~1s later stop. Since then I can disarm/arm multiple times and motors won't start again. Moreover, when entering the motors tab in configurator they don't spin. Before first arm the motors tab work perfectly ok in DSHOT.

I'm using Wraith AIRBOT 35A ESCs, however the problem arises also with other ESCs

# status
System Uptime: 118 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 15.37V (4S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, BARO=BMP280
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
SD card: Manufacturer 0x1b, 976896kB, 10/2011, v1.0, '00000'
Filesystem: Ready
Stack size: 6144, Stack address: 0x20010000, Heap available: 1612
I2C Errors: 45, config size: 5901, max available config: 32768
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 8, cycle time: 254, PID rate: 3937, RX rate: 110, System rate: 10
Arming disabled flags: CLI
VTX: not detected

----------------------------------------------------------------------------

# version
# INAV/KAKUTEF7 2.5.1 Jun 16 2020 / 12:11:32 (098a574b3)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

smix 0 0 18 100 0 -1
smix 1 1 18 100 0 -1

# servo

# logic

# gvar

# gf

# feature
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 2 4 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 4 1425 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# master
set looptime = 250
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 115
set gyro_lpf_type = PT1
set gyro_stage2_lowpass_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 200
set dynamic_gyro_notch_min_hz = 140
set acc_hardware = MPU6000
set acczero_x = 56
set acczero_y = 6
set acczero_z = -47
set accgain_x = 4074
set accgain_y = 4093
set accgain_z = 4059
set mag_hardware = NONE
set baro_hardware = BMP280
set pitot_hardware = NONE
set min_check = 1050
set motor_pwm_rate = 2000
set motor_pwm_protocol = ONESHOT42
set throttle_idle =  12.000
set model_preview_type = 3
set applied_defaults = 2
set gps_provider = UBLOX7
set gps_sbas_mode = EGNOS
set mc_airmode_type = THROTTLE_THRESHOLD
set nav_extra_arming_safety = OFF

# profile
profile 1

set mc_p_pitch = 26
set mc_i_pitch = 60
set mc_d_pitch = 28
set mc_p_roll = 22
set mc_i_roll = 50
set mc_d_roll = 24
set mc_p_yaw = 44
set mc_i_yaw = 70
set dterm_lpf_hz = 110
set dterm_lpf_type = PT1
set dterm_lpf2_hz = 170
set dterm_lpf2_type = PT1
set use_dterm_fir_filter = OFF
set yaw_lpf_hz = 30
set dterm_setpoint_weight =  0.850
set d_boost_factor =  1.500
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set rc_yaw_expo = 70
set roll_rate = 85
set pitch_rate = 85
set yaw_rate = 75

# battery_profile
battery_profile 1

# end the command batch
batch end

[LOG00016.TXT](https://github.com/iNavFlight/inav/files/4874861/LOG00016.TXT)

DzikuVx, 2020-07-05:
> with a confidence of 0.96. Please mark this comment with 👍 or 👎 to give

https://youtu.be/hEx174fj12E

DzikuVx, 2020-07-05:
Also, Oneshot42 is not recommended as very little ESCs support it. Switch to Multishot or DSHOT

closed 2020-07-05 by DzikuVx