iNavFlight/inav #5911: Pikoblx servos fail to init

unitware, 2020-07-05:
## Current Behavior
I get an error when initializing the PWM module, hence there is no servo output on the pins. Motor is working though.

Excerpt from cli status with added printf:
````
PWM output init error: Not enough servo outputs/timers
  getServoCount: 4
  timOutputs.maxTimServoCount: 3
````

The "enable moteor and servo" is enabled, and airoplane is selected in the mixer tab.
 
## Steps to Reproduce
Just flash and configure a pikoblx board and check the servo outputs

## Expected behavior
I'd expect that the servos should move when the sticks are moved.

## Suggested solution(s)
I suspect that there is a conflict between selected timers and other resources, I'd love som pointers to documentation or other hints so I can make this work.

## Additional context

Full Cli status with added printf, and a diff:
````
# status
System Uptime: 14 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6000, ACC=MPU6000
STM32 system clocks:
  SYSCLK = 72 MHz
  HCLK   = 72 MHz
  PCLK1  = 36 MHz
  PCLK2  = 72 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=NONE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE
Stack size: 6144, Stack address: 0x10002000, Heap available: 904
I2C Errors: 0, config size: 4673, max available config: 6144
ADC channel usage:
   BATTERY : configured = ADC 2, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = none
  AIRSPEED : configured = none, used = none
System load: 4, cycle time: 1014, PID rate: 986, RX rate: 49, System rate: 9
Arming disabled flags: 0x8140000
PWM output init error: Not enough servo outputs/timers
  getServoCount: 4
  timOutputs.maxTimServoCount: 3

# diff
# INAV/PIKOBLX 2.5.1 Jul  5 2020 / 16:42:26 (098a574b3)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

mmix reset

mmix 0  1.000  0.000  0.000  0.000
smix reset

smix 0 3 0 100 0 0
smix 1 4 0 100 0 0
smix 2 3 14 100 0 0
smix 3 4 14 -100 0 0
smix 4 5 2 100 0 0
smix 5 2 1 100 0 0
feature MOTOR_STOP
feature PWM_OUTPUT_ENABLE
serial 20 32769 115200 115200 0 115200
serial 0 0 115200 115200 0 115200
serial 1 1 115200 115200 0 115200
serial 2 64 115200 115200 0 115200
set acc_hardware = MPU6000
set acczero_x = 24
set acczero_y = -4
set acczero_z = -40
set accgain_x = 4085
set accgain_y = 4058
set accgain_z = 4110
set receiver_type = SERIAL
set rssi_channel = 14
set serialrx_provider = IBUS
set platform_type = AIRPLANE
set model_preview_type = 14
set small_angle = 180
set applied_defaults = 3
set ledstrip_visual_beeper = ### ERROR: VALUE 148 OUT OF RANGE FOR ledstrip_visual_beeper

set log_topics = 4294967295
profile 1

set yaw_lpf_hz = 30
set rc_yaw_expo = 70
set roll_rate = 30
set pitch_rate = 15
set yaw_rate = 9
battery_profile 1
````

---
Built by me for debug info, same error in the 2.5.1 release
# version
# INAV/PIKOBLX 2.5.1 Jul  5 2020 / 16:42:26 (098a574b3)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

unitware, 2020-07-05:
Further testing shows that servos start to work if I disable the ADC.
I'll dig into this and try to find out if the vbatt measurement is configured correctly, I'd like to use that

unitware, 2020-07-05:
@shellixyz I see that you are one of the few that has used this hardware and know the internals. Did you use it in a fixed wing and did the ADC work for you at that time? Can you test it again with 2.5.1 just to rule out any beginner mistakes on my side with the configuration?

unitware, 2020-07-05:
ok, further investigations revealed that there was a pin conflict.

The current sense ADC was set to PA2, and so was the 6th pwm output. 

I guess that it could be used as a analog input for an external current sensor board but I do not think that it should be setup like this in the target.c as default. Also the board does not have a rssi pad and that was also defined.

I'll create a pull request for this for examination.

DzikuVx, 2020-07-06:
as we have a fix, thanks a lot!

closed 2020-07-06 by DzikuVx