iNavFlight/inav #5923: STALL indication for fixed wing on OSD, when airspeed is below predefined value

iromadm, 2020-07-07:
## Current Behavior
Not supported

## Desired Behavior
OSD to show a warning message "STALL" as a separate OSD item when Airspeed drops below the user-configured value

## Suggested Solution
1. Add OSD item "Airspeed Stall"
Show only when Airspeed sensor detected
2. Add the "Airspeed Stall" configuration parameter with the integer field. Help text: "User-defined parameter of Airspeed. To be determined experimentally - the lowest speed when the aircraft starts stalling. This will trigger a message on the OSD"
Show only when Airspeed sensor detected
3. Show OSD Item text "STALL" when Airspeed drops below Airspeed Stall configuration parameter

## Who does this impact? Who is this for?
Fixed-wing models with Airspeed sensor installed.

## Additional context
I can write documentation and provide the necessary support for the developer who can take this

noagendafpv, 2020-07-23:
It would be great to have this feature for speed below a defined minimum as well as above a defined maximum. 

The max speed would be good for models like the Drift where Pawel had issue with over speed causing spin of death and crash.

MartianLocust, 2020-07-23:
You could be super-pimp and make it a "Dynamic Stall" warning, taking into consideration acc[Z].  Say your stall speed is Vs in straight and level, 1g flight. The dynamic stall speed would be (acc[Z] Vs^2)^1/2.  That still works at 1g, but also warns you when youre approaching the Clmax at any given dynamic pressure. 

As you're using the airspeed indicator, this intrinsically has density correction inbuilt, so no need for any other altitude or temperature correction.

iromadm, 2020-09-21:
> You could be super-pimp and make it

I would rather start from at least STALL indicator, and then start using airspeed for autonomous flights

Flight-instructor, 2020-12-22:
Nice feature, but htat makes only limited sense, as the stall is not pending on a specific speed, but on the angle of attack.
Unfortunaltel we mostly do not know at wht angle of attack our wing stalls.

Limited sens also means, there is a sense. Du to the fact that we mostly have an aircraft at the same weight when powered electrically, the stall angle of attack is reached at a specific speed. We an experimentally find out, what speed this is.

and for turns, there was a recommendation to help out with the measured G-force. Well done , Martian.

So, to be honest, I also would like to have such feature.

Be prpared to set the staqllwarning not at the stall speed, but about 10 to 15km/h or 10 knots above that speed to have enough remaining time to recover before spinning down:-)

sparkplug23, 2023-06-05:
Any progress on this? I would really like to have an alarm for slow speed, since when coming into land I want to get as slow as possible without getting below what I know is my aircrafts stall speed. A minimum alert would just help make it easier to slow down/fly without actually tracking the speed with my eye in FPV.

b14ckyy, 2024-03-20:
We have Min and Max airspeed alarms as linked. 

A Stall by its definition is nearly impossible to detect without dedicated angle of attack sensors and very specific knowledge about your airframe. This was explained already in this thread. I think this can be closed @DzikuVx

closed 2024-03-20 by sensei-hacker