iNavFlight/inav #5931: iNav 2.5.1 Matek F411 Wing Mixer Issue

mrduncan626, 2020-07-10:
When setting up the Mixer on a Matek F411 Wing (flashed with V2.5.1) as a Nano Talon V-tail, servo’s 3 and 4 are mapped to 4 outputs (S4, S5, S6, and S7).  

Please advise on how to connect the elevon servos to the FC.

[image]

jreise-d, 2020-07-11:
Same here. It seems to be a bug in the configurator, inav runs without an error. See issue in the configurator page https://github.com/iNavFlight/inav-configurator/issues/913

mrduncan626, 2020-07-11:
Same bug is covered under the configurator issues page as noted above.  Closing this issue.

closed 2020-07-11 by mrduncan626