iNavFlight/inav #5937: When erasing black box log blank screen

JCFPV, 2020-07-11:
When connected to Inav 2.5 when I delete blackbox log the screen says may take up to 20 seconds then once you click confirm the small box goes blank and doesn't erase the log without letting it stay blank for a few minutes maybe less then clicking cancel and power cycling the flight controller. If I do that it does erase the BB log just has to be power cycled 
[image]

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

kilrah, 2020-07-15:
Sorry to derail, but is there something special you needed to do to make the flash work on this board? 
Mine always says no chip found, tried both with stock FW, the FW distributed by Matek, both 2.5.0 and 2.5.1 from INAV configurator, and even tried BF for good measure and it's never seen.

I guess it's hardware failure but the seller tells me "they're pretty sure the hardware is OK"...

[image]

sammichael25, 2020-07-18:
I can also confirm this error.

FC Board name and vendor: Kakute F4 v2
INAV version: 2.5

amfern, 2020-07-21:
Same problem here, here is the log from configurator, i can provide more if needed
```
MSP data request timed-out: 8192
script.js:45204 MSP data request timed-out: 113
script.js:45204 MSP data request timed-out: 151
script.js:45204 MSP data request timed-out: 8192
script.js:45204 MSP data request timed-out: 113
2script.js:45204 MSP data request timed-out: 8194
script.js:45204 MSP data request timed-out: 151
script.js:45204 MSP data request timed-out: 8192
script.js:45204 MSP data request timed-out: 113
script.js:45204 MSP data request timed-out: 8194
script.js:45204 MSP data request timed-out: 151
script.js:45204 MSP data request timed-out: 8192
script.js:45204 MSP data request timed-out: 113
script.js:45204 MSP data request timed-out: 8194
2script.js:45204 MSP data request timed-out: 151
script.js:45204 MSP data request timed-out: 8192
```

stronnag, 2020-07-21:
This is a configurator bug and should be moved to the configurator repository issue tracker.
Erasing flash from the CLI or mwp's `flashdl` tool (which uses MSP) works OK.

amfern, 2020-07-21:
The issue is while erasing the flash, MSP messages don't work until it complete.

amfern, 2020-07-21:
The issue is while erasing the flash, MSP messages don't work until it complete.
I copied the flashing logic from betaflight, i believe this is the appropriate fix for on the FC side  #5964

on the configurator side the progress animation is missing.

sammichael25, 2020-07-25:
> This is a configurator bug and should be moved to the configurator repository issue tracker.
> Erasing flash from the CLI or mwp's `flashdl` tool (which uses MSP) works OK.

Whats the command to erase blackbox from cli?

stronnag, 2020-07-25:
Too difficult to enter the cli and type `flash<TAB>`? 
```
# flash<TAB>
flash_erase	flash_info	flash_read	flash_write	
# flash_erase 
Erasing...
# exit
```

stronnag, 2020-07-25:
otoh, we should document this somewhere obvious. PR to follow.

amfern, 2020-07-29:
Made a fix for the missing animation animation https://github.com/iNavFlight/inav-configurator/pull/1045