iNavFlight/inav #5939: Motor remapping

undefine-man, 2020-07-12:
## Current Behavior

can't remapping the motor number.
## Desired Behavior

## Suggested Solution

add this feature like Betaflight
## Who does this impact? Who is this for?

all users install esc unconventionally or just need this feature.
## Additional context

I once used betaflight , and my ESC was installed backwards for some reasons and i didn't make any adjustment to the cable ,cause betaflight can remap the motor number in CLI. But i fond it doesn't support this function after i switch to inav .So i hope this new feature can help inav more friendly to use .

DzikuVx, 2021-04-07:
INAV Configurator 2.6 has a Motor Wizard that does the same by modifying the mixer.

closed 2021-04-07 by DzikuVx

Tep-S, 2022-01-28:
@DzikuVx why there no such option for airplane? I can't use second motors as servo, can't use led pin as servo.
Just option in cli would save a lot of time and money.

Pavel-trying, 2022-03-18:
> INAV Configurator 2.6 has a Motor Wizard that does the same by modifying the mixer.

It not works, motors starts at the same positions as before "wizard mixer"... How i can fix it?

DzikuVx, 2022-03-19:
Wizard modifies the mixer nit output order. Effect is the same

Pavel-trying, 2022-03-19:
> Wizard modifies the mixer nit output order. Effect is the same

Thanks man, i found your video about that right after typed it..
Thanks for your videos, i love you man!

Pavel-trying, 2022-03-19:
But another mapping problem, why i cant add 3-th servo?
It connected to S10 pin. How to do it?

Board Matek H743
[image]