iNavFlight/inav #5940: Full duplex softserial-1 for Matek F411-WSE

Target0815, 2020-07-12:
## Current Behavior
Softserial-1 is currently only a TX.

## Desired Behavior
Softserial-1 should be fullduplex RX/TX

## Suggested Solution
Fullduplex Softwareserial-1 as this is also generated by default as a target for Matek F411-Wing. Only a small change is necessary:

So far:
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_TX_PIN     PB9 // ST1 pad
#define SOFTSERIAL_1_RX_PIN     PB9

New:
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_TX_PIN     PB9 // ST1 pad
#define SOFTSERIAL_1_RX_PIN     PB10 // LED pad

This would give us 2 full UARTS (GPS, CRSF) and 1 Softserial e.g. for the DJI FPV System (OSD). 

## Who does this impact? Who is this for?
Alle Benutzer mit DJI FPV-System auf Kopter und Flugzeug. Or of course all users who need a 3rd serial port.

satsepp, 2020-07-13:
would be nice to Control camera device although.

Sepp

fchansa, 2020-07-13:
Full ack!
For use of gps/crossfire/dji air unit it's neccessary to have a softserial with fullduplex RX/TX!

gregm412, 2020-07-19:
Does anyone have a build with this enabled? I found one at https://fpv-community.de/threads/matek-f411-wse-softserial.85747/  but I'm unable to download.

Target0815, 2020-07-19:
Download from this post: https://fpv-community.de/threads/matek-f411-wse-softserial.85747/post-1051752 (may be, that you have to be registered at the forum).

Feldsalat, 2020-07-19:
Here: [2.5.1](https://workupload.com/file/ThpWwcXQ9JT)

tobiasbischoff, 2020-08-02:
Please do it!

dl1bg, 2020-08-10:
My F411WSE is also waiting to be used with GPS + Smartport + Caddx Vista. Would highly appreciate this feature!

dl1bg, 2020-10-22:
@Feldsalat 
Tested, works fine.
I do not have enough knowledge to make a pull request by my own. But if anyone does, I will be happy to test it, brew the coffee or spend a beer.

alexvazquez, 2020-11-15:
Is this a new firmware that I can flash in INAV? I need to use my Caddx vista

alexvazquez, 2020-11-27:
> My F411WSE is also waiting to be used with GPS + Smartport + Caddx Vista. Would highly appreciate this feature!

Did you make it work? Appreciate it

chr0mcom, 2020-11-27:
i need this for dji vtx

dl1bg, 2020-11-30:
My EFX Racer plane works very nice with INAV; automatic flaps, RTH on failsafe and OSD is really usefull.
Unfortunately, there is no space for a bigger flight controller than the F411, so I can use this one or nothing.

UART1: GPS (RTH, speed measurement)
UART2: Receiver (FPORT in my case)
SOFTSERIAL1: DJI FPV system
I do not really think that someone would buy full color LED for the price of DJI OSD or ESC telemetry.

Still using the build from @Feldsalat, but it would be really kind if someone could spend the time to make a pull request. As stated above, I would be happy to brew him the coffee or spend a beer.

alexvazquez, 2020-12-04:
> My EFX Racer plane works very nice with INAV; automatic flaps, RTH on failsafe and OSD is really usefull.
> Unfortunately, there is no space for a bigger flight controller than the F411, so I can use [...]

Nice, basically same build I need, where did you connected the DJI FPV RX and TX? Led pin and what other?

dl1bg, 2020-12-20:
RX from DJI (Caddx Vista in this case): ST1
RT from DJI (Caddx Vista in this case): LED

alexvazquez, 2020-12-20:
> @alexvazquez
> RX from DJI (Caddx Vista in this case): ST1
> RT from DJI (Caddx Vista in this case): LED

@dl1bg Does the firmware is available to flash via INAV or I have to download from someplace the HEX file? Appreciate it

thedoble, 2021-01-10:
I would like this too! This board is great for small builds with gps/xfire/dji

jstoeffler, 2021-04-29:
This seems fixed, isn’t it?

https://github.com/iNavFlight/inav/pull/6353

alexvazquez, 2021-05-13:
Does anyone know where to get the firmware MATEKF411SE_FD_SFTSRL1 ? Appreciate so much

alexvazquez, 2021-05-13:
> This seems fixed, isn’t it?
> 
> #6353

Hi, did you get the HEX in order to test that OSD works on ST1 and LED pins?

jstoeffler, 2021-05-24:
I've tried with the V3.0-rc1: https://github.com/iNavFlight/inav/releases/tag/3.0.0 – inav_3.0.0_MATEKF411SE_FD_SFTSRL1.hex

And I can confirm it works

baddb1, 2022-04-29:
> I've tried with the V3.0-rc1: https://github.com/iNavFlight/inav/releases/tag/3.0.0 – inav_3.0.0_MATEKF411SE_FD_SFTSRL1.hex
> 
> And I can confirm it works

I just flashed with V3.0. What pin is for Tx and what is Rx? Thanks

jstoeffler, 2022-04-29:
> > I've tried with the V3.0-rc1: https://github.com/iNavFlight/inav/releases/tag/3.0.0 – inav_3.0.0_MATEKF411SE_FD_SFTSRL1.hex
> > And I can confirm it works
> 
> I just flashed with V3.0. What pin is for Tx and what is Rx? Thanks

@baddb1 If you read this thread you will find this information:

```
RX from DJI (Caddx Vista in this case): ST1
RT from DJI (Caddx Vista in this case): LED

```

bnmoore, 2022-05-26:
Does this work for HDZero? I can't seem to get it to work.

MrD-RC, 2022-05-26:
> Does this work for HDZero? I can't seem to get it to work.

No, HDZero needs an actual UART, not SoftSerial.

bnmoore, 2022-05-27:
Confirmed. HDZero worked on an actual UART. Is this a hard limit or just something that iNav or HDZero enforces due to performance?

closed 2023-03-06 by DzikuVx