iNavFlight/inav #5942: No RTH in OSD

FXvBH, 2020-07-13:
Hey
I have a small problem with RTH in INAV with fixed wing.
What do I have:
A Phantom FX-61 with Omnibus F4V3 FX.
I have flashed INAV on it, and for the flight modes I use anual mode and Horizon mode. This works well.
Now I have added RTH, but see that the OSD for RTH sets AIR mode.

So with manual mode - MANU
At Horizonmode - HORI
And by ReturnToHome - AIR

Have not tried it yet, but think it will not work either.

What am I missing?
What am I doing wrong?

Thanks, greetz FX

DzikuVx, 2020-07-13:
RTH activates only after arming and when home distance is > threshold. You can not "dry test" it just by flipping a switch on the ground

closed 2020-07-13 by DzikuVx

FXvBH, 2020-07-13:
Hi Powel.
Arming is working, and 14 sats. But I only tried to turn it on here in the garden. I now understand that there must be a "threshold" distance for RTH to do anything. What an embarrassment if it is so. This is the first time that I have done anything with an FC, including RTH.
Thanks for your quick reply, I will let you know if it solves this. If not, you will hear that too. Thanks again, greetz FX

PS, I think wednesday time for a test.

FXvBH, 2020-07-15:
Hi Pawel
1009/5000
I tested today. You were right. If I switch it to RTH now, "RTH" and "EN ROUTE TO HOME" will appear in the OSD. But.......................
If I fly off in a straight line and turn RTH on, it will reverse immediately, but it will hold that turn a bit. Sometimes counterclockwise, then a few times clockwise. It stays tidy and you hear it regulate the gas itself. I made a video with an HMDVR, and you see that the OSD neatly presents the RTH in the picture, and that it continues to perform its laps at height. Difference in height is about 2 to 3 meters, so that's neat.
I put RTH on 150 meters away from me, because otherwise I might not be able to see the FX_61 properly.

But why does it keep turning in circles like in Loitermode and drifting slowly downwind.

This is the first time that I have been involved in things like this, so I have to learn a lot.

Compass problem or whatever? I have no idea.

Mvg, FX

FXvBH, 2020-07-15:
The compass is from the GPS, and calibrated. I have results by compass check in CLI

FXvBH, 2020-07-21:
Hello
I have changed a number of things.
Pitch from 90 to 130 dps.
Roll from 350 to 250 pds.
Yaw PID to zero.
Compass off.

The FX-61 now flies in a straight line to the home position. And circles neatly at 70 meters above me.

I let Horizon flight mode exist. Needs 3.5 degrees pith, now flies neatly in horizon mode.

Groeten, Frans.