iNavFlight/inav #5943: Airbot OmniNXT f7

Cynicus, 2020-07-13:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

Cynicus, 2020-07-13:
Target OMNIBUSF7NXT
Airbot

file: target.h

Betaflight:
#define USE_UART2
#define UART2_RX_PIN            PA3
#define UART2_TX_PIN            PA2

INAV:
#define USE_UART2
#define UART2_RX_PIN            NONE
#define UART2_TX_PIN            PA2

UART2 is free on board. Why UART2_RX_PIN NONE?
UART1 reserved for smartaudio/trump.

With this configuration SBUS not work on UART2

digitalentity, 2020-07-14:
Which board specifically? OMNIBUSF7NXT target is used on a lot of clones.

Cynicus, 2020-07-14:
> Which board specifically? OMNIBUSF7NXT target is used on a lot of clones.

Original Airbot OmniNXT F7
https://store.myairbot.com/flight-controller/omnibus-f7/omninxtf7.html

Cynicus, 2020-07-14:
I made changes on my local comp and rebuild inav 2.5.1. All work fine!

closed 2020-10-04 by stale[bot]