iNavFlight/inav #5945: Use Craft Name to show informations using DJI FPV

FPVZaphod, 2020-07-14:
## Current Behavior
The DJI FPV OSD cannot show iNav warnings, it's currently not supported. But it can show a user defined craft name.

## Desired Behavior
Would be nice to have the option, to "misuse" the Craft name to show Warnings. 

## Suggested Solution
One possibility could be to have a check box "Mirror Warnings to Craft Name". If it's possible to change the craft name in-flight, this should be easily possible. Place the Craft Name somewhere in the middle, and it will be populated with warnings, once they occur.
It might be even used to show values not supported by DJI, like 3D-GPS-Speed. 

## Who does this impact? Who is this for?
All Pilots using DJI Digital FPV

## Additional context
If there is a contact between iNav Devs and DJI, it might be even useful to ask them if they can provide a few more free text fields, similar to the Craft Name. Then this can be used for showing new flight modes, without the requirement to wait for Firmware Support of DJI.  In best case, even upload of custom characters by serial connection could be supported. They the OSD would be really flexible in use.

mf0o, 2020-09-29:
I started to look into this just for own interest. 
digitalentity wrote the MSP driver for DJI OSD nicely new from scratch. I'm fairly new to c++ but got some basics to work. 
Seems to me that DJI is shortening the craft name larger than 10 characters and appending some "..."
Like: "WAITING FOR GPS FIX" will become "WAITING FO..."

Unless DJI supports the warning with more characters, we have to rephrase and short each warning message. i.e. "NO GPS FIX"
I would try to proceed unless digitalentity totally mislikes it.

FPVZaphod, 2020-09-29:
Whow, cool - thanks!
For the most important warnings, 10 characters should be ok.

mf0o, 2020-09-30:
So I got most of it working, as proof of concept, by duplicating the code from osd.c into osd_dji_hd.c. 
Seems to work well, even with multiple warnings and  OSD_ALTERNATING_CHOICES.

Maybe @digitalentity (and/or @DzikuVx) could give his input on the overall idea, using the DJI_MSP_NAME for warnings as a workaround, instead of waiting for DJI and a proper warning support.

I would welcome this, although my approach isnt "worthy" for sure.

[example: https://www.dropbox.com/s/vxw11l17gflujkj/inav_craft_name_to_warnings.MOV?dl=0 ]

FPVZaphod, 2020-10-01:
The craft name is anyways 100% useless in DJI HD. OSD content is not shown in ground recording or air unit recording, only in the goggles. Usually I know which plane I am currently flying.
Great work, video looks absolutely promising!
Maybe it's even possible to show a choosen value while there is no warning present? Heading would be great... Or mAh/KM efficiency :–)

mf0o, 2020-10-03:
I did some more adjustments to the messages and tested various types like AUTOTRIM and AUTOTUNE, AUTOLAUNCH etc. 
Also tried to replace the craft name in general with trip distance. Unit tests passed but havnt flown yet due to weather. 
Wanna give it a try? What target would you need?

FPVZaphod, 2020-10-03:
Trip Distance is cool, for later maybe we can collect some things to choose between?
Heading can be important,  mAh/km can be useful for tuning, 3D GPS Speed also something missing in DJI.
I use FuriousFPV F-35, or Omnibus F4 Pro also fine!
Thanks for the great work, I hope we see some reaction of Digitalentity!
Weather is stupid here for days, proper test subject would be a submarine :-/

mf0o, 2020-10-03:
! use with caution: [F35](https://www.dropbox.com/s/e80pllfjk5fvn76/inav_2.5.2_FF_F35_LIGHTNING.hex?dl=0) & [OmnibusF4](https://www.dropbox.com/s/6lg9m78ji9yasfe/inav_2.5.2_OMNIBUSF4.hex?dl=0)

FPVZaphod, 2020-10-10:
Still stupid weather here, but on the Bench it works great!
Man, this is a phantastic addon, and it deserves to be in the next iNav Version. How can we accelerate this?
This cannot be ignored from the Devs, it´s a HUGE improvement!
Based on that, it would be great if DJI could give a few more free text fields. Makes life way easier for them, because then it's a flight controller software thing to decide what to show. Imagine, with a few more characters you could even do something like:
.....N.....E..    <= Compass heading
@digitalentity : PLEASE, take a look at this!

mf0o, 2020-10-10:
I was just to update this.. :)
I flew almost 50km with that code since my last post and also „published“ it on my fork. 

For me, warnings and tripDistance work great. Now I know that my 4S digital Dart250 will carry me 15km on a single pack easily. 

Ive talked to Konstantin in the telegram chat about this and he seem to like the idea. 
Not sure if hes able to squeeze this in 2.6 or prefers a pull request by me. 

However, I will continue and refine the messages and to implement a cli option to choose the default entry as well as iterating multiple.

FPVZaphod, 2020-10-10:
Great, this would be perfect!
Efficiency in mAh/km is available in the normal OSD, so maybe it’s even ready to use in a variable and doesn’t need to be calculated. 
Fir CLI, maybe a bit to dec encoded value for the permanent cycle?
1 = trip distance
2 = 3D ground speed
4 = heading
8 = thr / autothrottle value

This would not allow the order of the valued displayed, but I don’t think this is a need.
Maybe another value for the duration how long to show each value before changing to the next.
Great work, I demand better weather!!

singlag, 2020-10-15:
> ! use with caution: [F35](https://www.dropbox.com/s/e80pllfjk5fvn76/inav_2.5.2_FF_F35_LIGHTNING.hex?dl=0) & [OmnibusF4](https://www.dropbox.com/s/6lg9m78ji9yasfe/inav_2.5.2_OMNIBUSF4.hex?dl=0)
> 
> edit: I miss the trip distance most, mAh/km could be calculated off that in theory. Without the trip distance its hard, maybe impossible, to properly plan a flight in case you want to go max range.
> But I like the idea of having it configurable. Maybe even multiple values that rotate, like the warnings where we have currently 5 iterating for 1 second.

can you make hex file for MatekF405SE (MatekF405 wing) ?
also, can you use "YAW PID" to show throttle level ?

mf0o, 2020-10-16:
Im working on an option to enable warnings, trip Distance, 3D speed, heading, throttle. If warnings are enabled, they will overrule the other elements. Each other element will iterate each 3 seconds or so.

Here a [MATEKF405SE.hex](https://www.dropbox.com/s/s3uc8ypv2yc0d1k/inav_2.5.2_MATEKF405SE.hex?dl=0 )of the previous code (tripDistance+Warnings in craft_name).
Any feedback welcome :)

singlag, 2020-10-16:
> Im working on an option to enable warnings, trip Distance, 3D speed, heading, throttle. If warnings are enabled, they will overrule the other elements. Each other element will iterate each 3 seconds [...]

thx, will try it outside soon

mf0o, 2020-10-17:
I got this a little further. Already had a new cli option implemented and then decided that this may be stupid if we get a proper field in DJI-MSP and have to remove the option again.
So I used the craft_name itself, to "program" the field. You can still use the craft_name as intended (and useless) but once you set the name to a 5bit-binary-name (00000 to 11111) you can enable/disable various features. 
This seems very hacky but it has zero footprint in any other file than osd_dji_hd.c

Available elements are: warnings, tripDistance, (auto-)throttle, 3D Speed, mAh/km
So i.e. "11000" would enable warnings and tripDistance.

Each element will shown for 3 seconds - if no warning is present. They will overrule the custom elements.
Here another [demo video](https://www.dropbox.com/s/5oqviz94s50wh33/inav_5bit_craft_name.mov?dl=0) where I rename the craft from "digitalDart" to "11111", turn on the transmitter, move the throttle.

FPVZaphod, 2020-10-17:
Amazing, nice solution! I like the way you choosed. No CLI required, every version of the iNav configurator will work. Like a little easteregg inside the OSD config.

singlag, 2020-10-19:
cool, OSD is much useful for DDJI fpv now
I suggest to show "throttle + 3D Speed" / "tripDistance + mAh/km" and shown for 1-2 seconds may be better ?

mf0o, 2020-10-19:
Next iteration. I did not like the way it was working and by chaning the expected "name" value, I could simplify the code a lot.

The name parameter has to be set to ":WDTSE" 
- ":" does enable the feature
- W to enable the warnings
- D for trip **D**istance 
- T for **T**hrottle (manual or auto)
- S for 3D **S**peed
- E for **E**fficiency mAh/KM

Each feature can be enabled, disabled reordered as preferred. i.e. ":SE" would work or ":WT" etc.
Guess I'm close to filing the PR ;)

FPVZaphod, 2020-10-19:
Whoa, so it's even possible to change the order - great work! 
Oh yes, PR and go! This is THE reason to go for the next update!

mf0o, 2020-10-20:
@FPVZaphod & @singlag 
Here are hex files of the PR code for you to test&run, more details on the usage are in the PR.
[F35](https://www.dropbox.com/s/e80pllfjk5fvn76/inav_2.5.2_FF_F35_LIGHTNING.hex?dl=0)
[OMNIBUSF4](https://www.dropbox.com/s/6lg9m78ji9yasfe/inav_2.5.2_OMNIBUSF4.hex?dl=0)
[F405SE](https://www.dropbox.com/s/s3uc8ypv2yc0d1k/inav_2.5.2_MATEKF405SE.hex?dl=0)

mf0o, 2020-10-24:
Ive been flying the latest master (v2.6) which also includes some adjustments by Konstantin on the DJI code without any issues. 
The mAh/Km does show the current consumption and not the overall efficiency like shown on the summary screen. 
Should I add the accumulated value?

Secondly, any RTH will override any values with „en route to..“ and „loitering..“. That was a little bit annoying as well.

FPVZaphod, 2020-10-24:
mAh/KM can be shown in the analog OSD, I don't have any idea if this is available somewhere directly. 
Would be same like Current divided by Speed I guess? 
Currently, here we have the worst situation. Corona (no people out bothering) BUT rain. Cannot wait to test this!

chrissng, 2020-10-31:
[Here's](https://www.dropbox.com/s/1399uplzosqpaz3/inav_2.6.0_MATEKF722SE.hex?dl=0) a build for the Matek F722-SE flight controller in case anyone wants to try it out.

Build from the last commit https://github.com/iNavFlight/inav/pull/6230/commits/5e4987179f12f9463f8d84049fbab814fb03a25e

mf0o, 2020-11-02:
Great, more testing is always good. Good is also that Konstantin added this to the 2.6 milestone.
Question is to you guys, any observations? Any item you would like to get added? Cosmetic corrections?
Not sure how much time is left and what can be done, but now is the chance :)

FPVZaphod, 2020-11-15:
I hope I'm not too late :-) What's about Crossfire Link data? Any chance for this? 
RSSI we have already, but Link Quality (LQ) is missing in the DJI OSD :-)

ArmyKnifey, 2020-11-23:
Hi Guys,
Just wanted to pass on my thanks for this awesome addition! I've recently invested into DJI and this was definitely needed.

mf0o, 2020-11-23:
> I hope I'm not too late :-) What's about Crossfire Link data? Any chance for this?
> RSSI we have already, but Link Quality (LQ) is missing in the DJI OSD :-)

Hey. It was already merged at that time if I remember correctly. 
Anyway, we can still extend the feature as it is in now (and build iNav ourself until official release).
Crossfire link data is a neat idea but the "RSSI" we get in DJI is already LQ, but we could add the new 1:100 2:100 format, or stuff like SNR.

DzikuVx, 2020-12-11:
#6230

closed 2020-12-11 by DzikuVx

ATKCA, 2021-03-02:
This thread is an example of how to work with the issue at hand and make progress. I linked the Betaflight devs but unfortunately they seemed uninterested and referenced a small change to the MSP code that will let DJI add warnings and it's up to DJI to implement it and if DJI don't then that's the end of it. 
Where as here you have rolled out your version in time for iNav 2.6 and anyone using it can run their DJI headset with confidence and enjoy lots of missing OSD elements. 
I have to be honest it's making a strong case to switch to iNav even if it means losing a little tuning performance vs Betaflight. 

I have a suggestion to help add more missing elements even the artificial horizon

• Change the position of the craft_name OSD element when displaying different values and if update rate supports it cycle it quickly so persistence of vision makes multiple readouts visible seemingly simultaneously. 

• I notice other elements that are less used are present on the DJI OSD like temperature and roll, pitch, yaw degrees. Could these be repurposed to provide other values?

FPVZaphod, 2021-03-02:
Yes, indeed this was a great example for feature request handling in an open source project. Very happy with the result.
Generally I think, this would be the best way to for for DJI. Offering several free text fields, not just one. Then it would be a pure software thing what to use and what not. Probably even the option to upload some custom characters, like you can do on 16x2 Style LCDs. Anyone here with some better connections to DJI?
I often use roll and pitch degrees for initial tuning the plane. Fly level in Acro, and then adjust Board roll and pitch in flight, until it shows roll/pitch degrees zero. 
A nice alternative option for a usage of this two values would be elevon servo output.  To show the deflection (-100 to 100) of each servo. Useful to detect oscillations due to excessive P / I. Or to see if close to a stall in a stabilized mode.

mf0o, 2021-03-02:
Ive thought about the artifical horizon but I guess the limits within the 1 line, 10 character field are too tight for a horizon to work. Even if we would move the OSD element around while the horizon is moving.
I know that Scavanger is working on more improvements of the OSD usage, and also Pawel has some stuff in the pipe for v2.7 (changing source of Speed Element for example). 
I just hope that those workarounds wont piss off DJI - afaik they dont have iNav on the radar yet.

ATKCA, 2021-03-05:
Thanks FPVZaphod and mf0o. I don't have connections with DJI but I have been contacting them on a range of issues both on their forum, by email and by opening up a support ticket. I will see if I can get them to commit to anything. Yes servo output would be a great usage of those readouts. Yes the artificial horizon would be limited on one line and it's effectiveness while moving the position would depend on the update rate. I would think 5hz or faster would give you somewhat persistent vision. That's great to hear there is yet more in the works already. I doubt this would annoy DJI if anything it makes their product more popular. Who knows maybe DJI will notice iNav now and choose to talk to the team here. Especially after the Betaflight team clammed up over little things like not naming fast SBUS the DJI specific protocol name they asked them to. I foresee great things ahead for iNav.

pitchlynn, 2021-03-19:
I may have to ditch betaflight and put inav on my freestyle quad!  I stumbled on this craft_name work around and then found an experimental fork of betaflight. I built it for my target and it works but has a slight issue.  I know I'm running betaflight but I think it was merged in from inav and they don't support it some I'm going to ask here. The first time I power up it never shows the warnings, only "craft_name".  But on subsequent power cycles it works.  I don't know what the difference is when it sits.  Only think I can think of is the battery on the GPS and that it gets a fix almost instantly when it hasn't been off for long.  Have you guys seen a problem with inav were you only get "craft_name" and not the warnings?