iNavFlight/inav #5948: No configuration received within 10 seconds, communication failed

Momike-6059, 2020-07-14:
I unfortunately had a crash with Mamba f405 FC.  Now it won't connect the INAV.  Red LED comes on, but nothing else.  Is my FC toast?  Is there a way to check?

Thanks

Mateyhv, 2020-07-14:
Try reloading iNav throught DFU, pressing the boot button

Momike-6059, 2020-07-14:
No luck.  My 3.3 volt chip is getting super hot which makes me think my CPU is fried. is there another way to test it?

pupojmastro, 2020-07-14:
I have the same issue with a FF 35-Lightning. As far as I can tell there's not a boot button but jumper pins. Went through "lost communication with board" and still not connecting. Just the solid red light on the FC.

Mateyhv, 2020-07-14:
> 
> 
> No luck. My 3.3 volt chip is getting super hot which makes me think my CPU is fried. is there another way to test it?

If STM32 isfried there is not much to be done. Can be exchanged in theory but you need the right equipment and yet its not sure you can solve it. Sorry

closed 2020-09-09 by DzikuVx