iNavFlight/inav #5949: How to calibrate reversible brushed motor ESC correctly in INAV 2.5.1??

uranus36, 2020-07-15:
Anyone knows the correct procedure to calibrate the reversible brushed motor ESC with INAV 2.5.1 ???
  I'm trying to build a Rover with Omnibus F4V3 in INAV 2.5.1. The brushed motor ESC is Hobbywing QuicRun 1060 with Forward/Brake/Backward function enabled. 
   Even if I checked "stop motor at low throttle", "reverse motor" and setup the 3D throttle high, low and nutral with 1450, 1550 , 1500. The motor can move forward and brake  sometimes only. Besides, the motor will rotate with fully power sometimes when I switch "ARM" button off. It's very weird.
    Another strange thing is that even if I set my radio nutral to 1500 correctly and it shows at INAV receiver tab correctly. When I change the tab to output page, the motor column will show it's at 1467. Could someone help to explain it?

tourerjim, 2020-07-19:
Ive got the HW Quicrun 1025 and cant even get the motor to drive, Im using the Kakute F722 Mini FC with Inav 2.5

howels, 2020-07-25:
Exactly the same plan here - Hobbywing Quicrun 1060 with Forward/Reverse mode.  Using 2 of these to build a tank.

Have you tried removing the jumper to remove the "brake" function so it's only F/R instead of F/B/R @uranus36 ?  Brake can be a weird setting and seems to mean that the sudden Foward-Reverse transition is avoided so as to not strip gearboxes. Of course if we are controlling the transition with iNav then this can be managed entirely in the FC.

tourerjim, 2020-07-25:
I've since got mine driving with Inav, you need to keep normal throttle linear in you TX, then in Inav, Modes need to set arming switch, then to drive switch on esc place throttle mid stick then switch arm switch wait couple seconds for esc to initialise.
Also you can't use TX trim to control steer end pounts need to be done in Inav, outputs and adjust Mid, Min, Max to centre and end points of steer.
I also tried brushless motor and esc and still didn't work as the problem is an arming issue as mentioned here,  https://www.youtube.com/watch?v=obfc62VWhyw https://www.youtube.com/watch?v=7MAdgGkBkXs...

howels, 2020-07-26:
I'm seeing the "reversible motors" feature uncheck itself after I click save&reboot. Any idea why? Stuck with inav thinking that motors only run forward although the ESC lowest value is full-reverse and highest is forward. So testing motors in outputs tab means that they blast into full reverse when enabled.

tourerjim, 2020-07-26:
In Inav Configuration do not check 'reversable motor tab' as you already have it in your esc.
You do not see reversable throttle parameters in Inav, your 1060 esc sees your TX mid stick and is controlled throught Inav throttle signal even thoigh inav throttle range is 1000 to 2000, yiu esc read the 1500 as neutral.

howels, 2020-07-26:
> In Inav Configuration do not check 'reversable motor tab' as you already have it in your esc.
> You do not see reversable throttle parameters in Inav, your 1060 esc sees your TX mid stick and is controlled throught Inav throttle signal even though inav throttle range is 1000 to 2000, your esc reads the 1500 as neutral.
> What your could do in Inav Mixer tab is to add new Mixer Rull leave Servo 0, Input as Stabilzed Throttle then you should see throttle action there as well as a reference.

But I want to be able to arm with throttle at 1500, unless I have reversible ESC I will need to arm with throttle at 1000 and ESC will go backwards!

tourerjim, 2020-07-26:
Unfortunately I'm working on mine and not able to look through the setup, when I set the arm switch to off and throttle low I then switch on esc, (2 bleeps) then arm switch on then mid stick and you should get a single esc bleep.
If you set reverse tab in configuration you don't get the bleeps for some odd reason.

howels, 2020-07-26:
I've got mine on the bench as well, I'll investigate further. Suspect there are some bugs with the motor mixer as it was re-factored for 2.6 AFAIK.

tourerjim, 2020-07-26:
If your able to send private measage from here with your email, im just staring to out mine back together once done i can send screen shots of my setup.

howels, 2020-07-26:
No PM options that I can see. Slowly getting the rover online. Next challenge is how I can use a motor for steering instead of a servo. I'm converting a SG1203 RC tank to inav.

tourerjim, 2020-07-27:
Just found this link could be helpful for you https://github.com/iNavFlight/inav/issues/5767

howels, 2020-07-28:
@tourerjim the rover is now fully operational. The hobbywing quicrun ESCs work great in fwd/rev mode (no jumper) and using the reversible motors setting with default deadband worked perfectly for me. I needed to set reversible motors via the CLI as the inav-configurator GUI kept reverting the setting but after that it worked perfectly.

tourerjim, 2020-07-28:
Just finished setting up the GPS and now dont have drive, what setting did you write and save into Cli

uranus36, 2020-07-28:
I have completed my build with F/B/R mode correctly. But the setting is really difficult and weird. Need to check the reversible motor box, then set nutral as 1467, high =1477, low =1457. I did set an ARM switch . In order to activate my Rover, need to power on the system without turning on ESC power, then turn on the ARM switch. Turn on ESC power and wait for couple seconds for ESC initialization. Then the Rover will move correctly. The weird thing is that even if I turn off the arm switch , it can still move forward....  Besides, if you don't enable brake mode, it will take a lot of time to stop the car when it's at high speed. I think it's kind of dangerous...

tourerjim, 2020-07-29:
Still cant get mine working since setting the GPS, if I uncheck 'reverse motors' and switch on esc I only get the two bleeps it's sensing throttle signal but no neutal point for 3rd bleep thoughout throttle range, if I set 'reverse motors' and then turn esc on i get the 3rd bleep no matter where the throttle range is so it seems when I turn on 'reverse motors' it cuts out the throttle signal, ive reflashed the board twice now and still no luck.

howels, 2020-07-29:
> Still cant get mine working since setting the GPS, if I uncheck 'reverse motors' and switch on esc I only get the two bleeps it's sensing throttle signal but no neutal point for 3rd bleep thoughout [...]

When you arm the ESC with a FC between receiver and ESC then the FC will supply your "neutral" deadband setting when the ESC initializes (unless you supply power to ESC after arming somehow). This means that the FC is setting your throttle mid-point for you. Use the inav-configurator outputs tab to see what's happening when you arm.

closed 2020-11-07 by stale[bot]