iNavFlight/inav #5960: Unable to arm "throttle not low" even when set

Ziov1, 2020-07-20:
I'm having an issue with being unable to arm due to throttle not being low. I've set it in receiver options and also in controller to the 1000 - 2000.

I also set it using set min_check = 1000 but still the same issue.

I don't seem to able to solve it, tried all the ways I did from searching online. 

Please help.
F722 se, x9d+.

DzikuVx, 2020-07-20:
THR has to be below `min_check`. But without diff or any extra data like a screenshow from the Receiver tab, we can not help

Ziov1, 2020-07-20:
Have attached files thanks.
[image]
[image]
[diff.txt](https://github.com/iNavFlight/inav/files/4949457/diff.txt)

DzikuVx, 2020-07-20:
`set min_check = 1050` should do it

Ziov1, 2020-07-20:
same issue

Ziov1, 2020-07-21:
Did a little more looking into it, not sure if related but magnomiter was not active, was getting i2c errors. They kept counting up. Activated magnomiter and i2c errors are at 45.
While magnomiter is active I can't get a GPS lock without it I get 16 Satalites. I'm using a Matek GPS M8q 5883.

Ziov1, 2020-07-21:
Still getting issue, sorted gps magnomiter.

Mateyhv, 2020-07-21:
> 
> 
> Still getting issue, sorted gps magnomiter.

Can you upload a video somewhere?

Ziov1, 2020-07-21:
> > Still getting issue, sorted gps magnomiter.
> 
> Can you upload a video somewhere?

What do you need video of?

closed 2020-07-26 by Ziov1