iNavFlight/inav #5961: Servo and motor not working on flying wing

ubervav, 2020-07-20:
Hello. I'm completely new to iNav and FCs. I'm currently building a flying wing with Matek F765 WING and iNav. I have followed YouTube tutorials and iNav's Fixed Wing guide, but when I want to test the servos and motors there's only one servo that is working (servo pin 3/left wing). The servo on pin 4/right wing will immediately go to full end point after plugging in the battery.
In iNav, I have enable the servos and motor and I can see that the FC is responding to my stick movements correctly.
After countless Google searches I have not found any useful tips to fix this problem.
Any help would be appreciated :D

DzikuVx, 2020-07-20:
We can not help without any data. CLI diff, Mixer tab screenshot. Whatever you can give us

ubervav, 2020-07-21:
Yes, of course. Here is a screenshot of the mixer tab. It's just the flying wing preset.

[image]

CLI diff:

[INAV_cli_F765_20200721_101500.txt](https://github.com/iNavFlight/inav/files/4952170/INAV_cli_F765_20200721_101500.txt)

Please let me know if you need more information.

closed 2020-07-22 by ubervav

PhilippeKonrad, 2020-11-05:
I am having the same issue on 2.5.2 for KakuteF4V2, only servo 4 isn't responding.

3 works flawlessly, but servo 4 boots, twitches, and goes to end point. 

Then no response from that servo, either via outputs tab or with tx. 

I ran the servo through a tester and it works fine. 

How did you resolve?

ubervav, 2020-11-10:
Unfortunately I didn’t resolve it. Nothing was working for me, no forums or help from others. I’ve been busy for a long time and haven’t been able to look at it for a while. I will have another look during the holiday seasons.
If you resolve it, please let me know. Maybe we could cooperate on this matter!