iNavFlight/inav #5966: Servo Trim to be selected in the Adjustments drop-down and OSD

stuwe911, 2020-07-21:
## Current Behavior
I find Servo Autotrim very difficult to use especially when you have to trim all 3 axis. It kinda works but not precise enough. It requires multiple flights to get that right so that these days i don't use it at all but do mechanical trimming instead. However for fine tuning I would like to do this in flight by selecting the axis and change step by step like any other Adjustments feature.

## Desired Behavior
Being able to select each axis in the Adjustments drop-down for Servo trim and show the values in the OSD

## Suggested Solution
Add the 3 axis (not each servo) to the Adjustments drop-down for selection. Also show the values in the OSD.

## Who does this impact? Who is this for?
Anyone flying Fixed Wing / Ariplanes / Rover / Boats

Avegawanderer, 2020-07-31:
I support this feature request a lot! There is a possibilty in OpenTX for custom use of trimmers. Using special functions it is possible to override adjustment channels. For example, I adjust cruise throttle by using throtle trimmer on my Taranis.

MrD-RC, 2020-08-13:
I just came on here to add this feature request. So it's a +1 from me!

Some LOS pilots are using iNav. Manual trims are much more natural for them, and would be easier to perform than servo autotrim. My thoughts, would be to disable the trims on the transmitter, and send them over the RC link to perform the adjustments in iNav. I know it would be 4 RC channels. But it would be natural to the pilot. The only difference would be needing to save when they land. It would be a great addition to iNav which may even pull more pilots in to using it.

INAV-FWG-Steve, 2020-08-13:
As MrD-RC mentioned, this would be a game-changer for LOS pilots.  The LOS community is our growth target for INAV fixed-wing over the next year.  We are looking to offer a turnkey solution to them as entry into INAV without FPV.

Avegawanderer, 2020-08-13:
I took a closer look at implementation. If we had a single servo for an axis, everything would be fine. We would adjust the middle position and display it on OSD. However, there are more than one servo for each axis (simplest example - two servos for roll axis). I see two possibilities: 

1. Adjusting every single servo (or, for example, first 8servos) and displaying it's middle position on OSD. Servos for same axis may be trimmed together if they use the same adjustment range. No dedicated trimmers, adjusting of each servo must be set up individualy.

2. Having some kind of dedicated "trimmer variable", one per axis. Those trimmer variables are adjustable, displayed on OSD, and may be applied at mixer as a rule.

I personaly prefer variant 2. Any thoughts?

MrD-RC, 2020-08-13:
I had a look too. I’ve not done a lot with iNav, so can’t say I understand how it all works or fits together. But it looks like there’s an array of the active servos, and some definitions for elevator, flaperon1, flaperon2, and rudder. Those look be keys to the array, or servo mutable. I’m guessing the flaperons are the ailerons in a plane setup. I guess also that a wing would use the elevator and 1 of the flaperons. I don’t really know. The other issue I guess is the weight, as some are reversed in the mixer (negative weight). Maybe the weight doesn’t matter, I don’t know at this stage. It would be easier if it doesn’t. I think I need to look more in to it and try and understand what is happening. I don’t understand how multiple elevator servos work with this for example. 

I don’t think the servo mid point should be applied in the mixer. It’s more an outputs feature, when you look at transmitters too. In the mixer, you’d be more looking at a curves type implementation. But they’re not used for trimming. 

My ultimate implementation:
1. Don’t care what’s on the OSD. But having the values may be useful for FPV, or just completeness. For the units, I think either the raw mid point in uS, eg 1456. Or base 0 as the mid point and have positive and negative percentages as suggested. 

2. The transmitter trims control the servo mid point for the related axis. So the yaw trim adjusts the yaw, roll the roll etc, no matter how many servos are on that axis. The first part is simple enough, just send the trim on the RC channel and assign that to the adjustment. The challenge is adjusting the correct servos in the correct directions.

3. Adding a logarithmic adjustment as opposed to using the current step adjustment. So at the moment you have to click each time you change a value. For the trims, it would be great if you could hold the trim switch in the direction. It starts adjusting slowly, but speeds up the longer you hold the switch. Just like trims on a transmitter. 

4. A min and max mid point range of 1375 to 1625. That’s about the same as a standard Tx trim adjuster. If it needs more than this, it should be changed physically, rather than extending the trim range. 

Just my thoughts, maybe they’ll stir some ideas.

Avegawanderer, 2020-09-29:
Similar issues:
Fixedwing Trims in Adjustments #5033 
Adding manual trimming of an aircraft #4132

FPVZaphod, 2020-10-02:
@Avegawanderer - well explained, makes me understand the problem much better.
While we enter a desired value before the mixer (Like in a conventional RC Radio), iNav saves the Trim per Servo - after the mixer. 
Possible solution would be to receive a new axis trim value (up / down in the Adjustments), then pause the PIFF Loop for a very short time (milliseconds), and save the new resulting servo output as midpoint. In this moment, only the trimmed input goes through the mixer, outputs new servo positions, and they are to safe as midpoints. Then the PIFF Loop will continue working, moving servos again. 

Maybe the whole Autotrim can be changed in the way it works. Instead of saving the control surface position required to keep the plane level, it might be possible to use the I-Term for doing the Job. If average I Output is Zero, then the plane is trimmed when flying level. If I is permanently positive or negative, the Plane is not trimmed.

Avegawanderer, 2020-10-02:
> Maybe the whole Autotrim can be changed in the way it works. Instead of saving the control surface position required to keep the plane level, it might be possible to use the I-Term for doing the [...]

Yes, that is a good idea. It may be better than current algorithm, but it needs testing. I guess, that's how it is implemented in Arduplane.
Still I personally prefer to have an opportunity to explicitly trim servos by adjustments. It would be useful for leveling an aircraft while flying in manual mode. I am thinking of making global variables (GVAR) adjustable. Instead of using dedicated trim variables, one may add a global variable to mixer rule if desired. Having an adjustable GVAR may be useful for many other purposes, not only trimming servos.

stuwe911, 2020-11-16:
> Still I personally prefer to have an opportunity to explicitly trim servos by adjustments. It would be useful for leveling an aircraft while flying in manual mode. I am thinking of making global [...]

I agree on that! lets optimize autotrim another time but add those GVAR for explicit trim. that would be very helpful to change a servo mid point in air! thanks!

stuwe911, 2021-11-11:
closing this issue now due to the new "permanently autotrim servos" in v3.0. it works very well and a manual trim during the flight is no longer required.

closed 2021-11-11 by stuwe911