iNavFlight/inav #5969: Remaining flight time and flight distance estimation Not Working

breadoven, 2020-07-22:
I've set up "Remaining flight time and flight distance estimation" according to the info in the Battery.md. However, I'm just getting dashes on the OSD, no numbers showing.

VBAT, CURRENT_METER and GPS features are enabled plus the following set:

set battery_capacity = 33300
set battery_capacity_unit = MWH
set nav_fw_cruise_speed = 1250
set cruise_power = 8200
set idle_power = 200
set rth_energy_margin = 5
set osd_estimations_wind_compensation = ON
set vbat_max_cell_voltage = 420

The battery has been Full when connected at the start of a flight (correctly detected in Configurator) with the number of cells specified in the battery profile to avoid automatic cell count errors (only 1 profile used). Last flight included a WP mission so all the Nav stuff is working fine.

So everything seems to be setup as required but no values are being displayed. What am I missing here ?

OMNIBUS F4 V3 FC running iNav 2.5.1 on a fixed wing.

breadoven, 2020-08-01:
Still no luck getting this to work. I guess from the lack of answers either:

- Few others use it
- Nobody can see anything wrong with my setup and therefore can't help
- I've made a glaring error I ought to sort out myself  :)

I did see in the code, rth_estimator.c,  mention of Launch mode:

if (!(feature(FEATURE_VBAT) && feature(FEATURE_CURRENT_METER) && navigationPositionEstimateIsHealthy() && (batteryMetersConfig()->cruise_power > 0) && (ARMING_FLAG(ARMED)) && **((!STATE(FIXED_WING)) || (**isNavLaunchEnabled() && isFixedWingLaunchDetected**()))** && (navConfig()->fw.cruise_speed > 0) && (currentBatteryProfile->capacity.unit == BAT_CAPACITY_UNIT_MWH) && (currentBatteryProfile->capacity.value > 0) && batteryWasFullWhenPluggedIn() && isEstimatedWindSpeedValid() && isImuHeadingValid()))
        return -1;

This wouldn't have anything to do with it would it (I don't use Auto Launch) ?

breadoven, 2020-08-01:
OK a quick test doing a "dummy" launch (never actually released it) in Nav Launch Mode which triggered the motor starting resulted in OSD Remaining Flight Time and Distance numbers being shown even without the aircraft actually flying. So these flight time and distance estimations only seem to work if you use Launch Mode. I can't see any obvious reason for this so is it possible to remove this requirement for those who don't use Nav Launch ? It should at least be mentioned in the Battery.md as a requirement to get this to work.

shellixyz, 2020-08-06:
I checked the code it is indeed a bug. Right now the remaining flight time/distance estimation is only working when the nav launch feature is enabled.

shellixyz, 2020-08-11:
@breadoven Could you give my fix a try ? Let me know if you need me to build a firmware image for you.

breadoven, 2020-08-13:
OK I'll give it a test and let you know if there are problems.

breadoven, 2020-08-17:
OK shellixyz I tried your fix and it's displaying correctly now without Nav Launch required.

Just need to improve the wind estimator now because it seems to cause problems with wildly inaccurate over-estimates sometimes. But I guess that's another issue.

shellixyz, 2020-08-17:
Ok thanks you @breadoven. I confirm the wind estimator never worked well. I suggest that you set `osd_estimations_wind_compensation` to `OFF` for now. It is necessary to improve the attitude estimation to make wind estimator work correctly and that's a complicated topic.

PR #6040 (2020-08-17): [RTH Estimator] Fix display condition
Fixes #5969
closed 2020-08-17 by shellixyz