iNavFlight/inav #5990: FF values can't be adjusted via OSD adjustments

b14ckyy, 2020-07-27:
## Current Behavior
with in flight OSD adjustments, P, I and FF values can be selected (blinking) but FF cannot be changed. For P and I it works fine. 

## Steps to Reproduce
1. set up adjustments for Pitch, Roll or Yaw PIDs (especially D) on a fixed wing
2. select the item you want to change, P, I and D (actually FF) blink when selected
3. try to change the value by the modifier channel
> P and I can be modified, D(FF) won't change

## Expected behavior
Possibility to change FF with adjustments

## Suggested solution(s)
fix the modifier 

## Additional context
Talked to Andreas K and he wanted to fix it. Would be nice to have that fix in 2.5.2 as we need that modification for a OpenTX community project. 
---

INAV/MATEKF411SE 2.5.1 Jun 16 2020 / 12:54:49 (098a574b3)

- FC Board name and vendor: All Boards affected. Tested on Matek F411WSE and Matek F405 Wing
- INAV version string:
INAV/MATEKF411SE 2.5.1 Jun 16 2020 / 12:54:49 (098a574b3)
GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

Pairan, 2020-07-28:
That goes hand in hand with [inav-configurator/issues/813](https://github.com/iNavFlight/inav-configurator/issues/813) I think

b14ckyy, 2020-07-28:
Yeah maybe but the FF value is shown as D in the OSD anyway as only one of both can be active. So on a plane the OSD shows PIFF and on copter PID. In OSD configurator it is still named as pitch PIDs and Roll PIDs but shows PIFF. 

the value does blink in OSD so it seems to get selected correctly but you may be right that the Plane mapping is missing there. Should be named Roll D/FF in adjustmenet selector instead of Roll D etc.

closed 2020-12-11 by DzikuVx