iNavFlight/inav #5994: Sbus Reciever and GPS taking up same UART Port

VTX802, 2020-07-29:
As the title says, I am unable to connect my GPS and receiver at the same time due to what I'm assuming is an error with the UART ports.  I am using [this Omnibus F4 Pro V3](https://www.racedayquads.com/products/omnibus-f4-pro-v3-flight-controller-with-current-sensor-osd-and-smartaudio?_pos=15&_sid=a60ae56c9&_ss=r) flight controller with a M8N GPS antenna and a X4R receiver connected with SBUS.  I flashed it with the OMNIBUSF4PRO version 2.5.1 firmware using iNav.  When configuring the ports, I found that only UART1 (Of the three options being UART1, UART3, and UART6) could be configured for the GPS or for the receiver, but none of the other ports work for either.  Based off the picture I included below, it looks like the port I connected my GPS into is hardwired as UART1 so is there a way to configure the other UART ports as serial ports for my reciever?  I'm open to any and all solutions needed to connect my receiver and GPS at the same time because at this point I feel like I've torn apart the internet looking for a solution.
Also, apologies if this post is off topic or breaks any rules here.  I'm new this forum and this software and haven't fully found my way around yet.  Thank you in advance for any and all help you can provide.

[image]

breadoven, 2020-07-29:
https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md

According to the above the SBUS inverter is on UART 1 so this can't be used for anything else when the Rx is connected to the SBUS pin on the board and the jumper set for SBUS. GPS should work on UART 6, if not maybe a board problem:
https://github.com/iNavFlight/inav/issues/3895

VTX802, 2020-07-29:
So I’ve accepted that the SBUS will take up UART1, however I’m still unable to connect the GPS to uart6. Is this something I could potentially fix with a custom firmware?

Thanks for the help

beehphy, 2020-08-01:
I am also experiencing this problem on my F4 Omnibus Pro V3. 
I am also using serial 1 as serial Rx, because sbus. (could not get PPM going after much effort).
My goal is connecting a Bluetooth module for PID tuning. Both Serial 3 and 6 are unsuccessful at MSP port setup.
On the bench, connect BT with a phone, sending dummy signals I can measure with an oscilloscope, the baud rate is 56K coming out of the BT just fine. 
Setting up Serial 3 and 6 was a total failure. I also attempted to set as telemetry outputs, but observed no signals on any of the 4 pins (Tx3, Rx3, Tx6, Rx6). I observe all but Rx3 toggling between GND and 5V when the FC reboots between configuration saves.

rakiata, 2020-08-03:
Hi, I used the Omnibus F4 Pro V3 from Aliepress
I flashed it with INAV/OMNIBUSF4PRO_LEDSTRIPM5 2.5.0
UART connections:
UART 1 - SBUS RX / UART 3 - COMPASS / UART 6 - GPS
After calibrating all sensors, I put the battery and the gyro in the sensor tab sits in a straight line when I move it. I went outside to find satellites and everything was fine until I arm it. Navigation is safe but the gyro does not want to work. I hope it helps with some advice
[dump 2.5.txt](https://github.com/iNavFlight/inav/files/5018221/dump.2.5.txt)
[status 2.5.txt](https://github.com/iNavFlight/inav/files/5018222/status.2.5.txt)

thegodfarmer, 2020-09-26:
i had the very same problem and found it only worked in ppm mode not sbus. my board was from banggood but my other boards of the same type from other sellers work flawlessly

closed 2020-11-26 by DzikuVx