iNavFlight/inav #5998: Max Altitude "GEO-Fence" Option

MartianLocust, 2020-07-30:
## Current Behavior
Should you have a runaway altitude event (like selecting NAV WP when WPs not loaded) an un-restricted climb may occur prior to pilot intervention. If operating in a sensitive area this is not ideal and may lead to airspace breach / operating outside the EASA/ANO(UK) privaledges.   

## Desired Behavior
Implement a iNav Configurator selectable MAX ALTITUDE limit that interrupts NAV mode and either holds at that MAX ALTITUDE or changes mode to RTH. 

## Who does this impact? Who is this for?
This is for FW and RW pilots to help avoid ANO/EASA altitude limit breachs if selected. For example I would select 400ft as MAX ALT. This may also help pave the way for science payload UAV manufacturers to utilize iNav as part of their package and prove compliance with new CE regs (C1 / C2 etc).

breadoven, 2021-11-29:
Assumed closed by https://github.com/iNavFlight/inav/pull/4442.

stronnag, 2021-11-29:
OTOH, it I read the title .....

closed 2021-11-29 by stronnag

breadoven, 2021-11-29:
Did think it was limited to altitude. Never going to be a true altitude geo fence without some topographic input given the nature of the rules when things aren't flat, but better than nothing.

stronnag, 2021-11-29:
> Did think it was limited to altitude. Never going to be a true altitude geo fence without some topographic input given the nature of the rules when things aren't flat, but better than nothing.

It was limited; mea culpa.
However, should you have a spare moment, there's always the stalled, locked (and in my simple view, somewhat over complex in implementation) #4798