iNavFlight/inav #6007: Implement sensor fusion/Kalman filters for inputs

skorokithakis, 2020-07-31:
There isn't really much to explain, Kalman filters are a great way to estimate current position based on sensor data, and I think this would be the single biggest improvement INAV could make right now. Has this been considered before?

fluffy, 2020-08-01:
I'm not sure but I suspect what you are taking about is already implemented in 
https://github.com/iNavFlight/inav/blob/master/src/main/flight/kalman.c

skorokithakis, 2020-08-01:
This seems to only work for the gyro and doesn't fuse gyro/accelerometer/GPS/inputs together to form the estimate. You should be able to get position estimates that are extremely close to true by fusing all the sensors together.

Some references:

https://robotics.stackexchange.com/questions/277/why-do-i-need-a-kalman-filter

https://www.bzarg.com/p/improving-imu-attitude-estimates-with-velocity-data/

digitalentity, 2020-08-01:
INAV uses Alpha-Beta filter for position estimation, which is pretty much identical to Kalman, except for the Kalman gain calculation. Tuning the parameters is easier for Alpha-Beta filter, which is also guaranteed to be numerically stable.

Note, that Kalman algorithm by itself is not a magic silver bullet, it requires a lot of supporting logic to produce sane results and handle edge cases (i.e. missing or misbehaving sensors). Implementing that supporting logic for existing code may provide more benefit than migrating to "proper" Kalman filter, therefore migrating to Kalman filter is not on the roadmap at the moment.

skorokithakis, 2020-08-01:
@digitalentity That makes sense, thank you for the reply.

closed 2022-02-10 by DzikuVx