iNavFlight/inav #6010: [Matek F411-WING] Remap Motor 2 to servo

adriancodre, 2020-08-02:
Hello, i tried to make a custom build and remap the motor 2 toa. servo to be able to use it on a headtracker. I tried changing:

DEF_TIM(TIM3, CH2, PB5,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S2  D(1,5,5)
to
DEF_TIM(TIM4, CH3, PB5,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S2  D(1,5,5)
or
DEF_TIM(TIM2, CH4, PB5,  TIM_USE_MC_SERVO | TIM_USE_FW_SERVO,  0, 0), // S2  D(1,5,5)

but the output does not do anything.

what timer to use and what channel ? the default config is:

DEF_TIM(TIM3, CH1, PB4,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S1  D(1,4,5)
    DEF_TIM(TIM3, CH2, PB5,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S2  D(1,5,5)
    DEF_TIM(TIM4, CH1, PB6,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S3  D(1,0,2)
    DEF_TIM(TIM4, CH2, PB7,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S4  D(1,3,2)

    DEF_TIM(TIM2, CH2, PB3,  TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0), // S5  D(1,6,3)
    DEF_TIM(TIM2, CH3, PB10, TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0), // S6  D(1,1,3)
    DEF_TIM(TIM2, CH1, PA15, TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0), // S7  D(1,5,3) - clash with S2

    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_ANY,   0, 0), //softserial_tx2 - 2812LED TIM_USE_LED   D(2,1,6)
    DEF_TIM(TIM5, CH1, PA0,  TIM_USE_PPM,   0, 0), //use rssi pad for PPM/softserial_tx1

thanks

Arvycka, 2020-08-03:
I don't think you can. Short answer - you can't use the same timer both for servo and motor (TIM3), and timers are assigned to physical MCU pins, so you can't use any timer for any pin. Read MCU datasheet, there is a table for what pins can use what functions.
Long answer: https://www.youtube.com/watch?v=v4R-pnO4srU

DzikuVx, 2020-08-03:
S1 and S2 have to have the same function assignment, not possible like @Arvycka wrote

closed 2020-08-03 by DzikuVx

adriancodre, 2020-08-03:
> I don't think you can. Short answer - you can't use the same timer both for servo and motor (TIM3), and timers are assigned to physical MCU pins, so you can't use any timer for any pin. Read MCU [...]

but i want to use TIM4 or TIM2 which are used for servo and leave de S1 on TIM4 alone which is used for motor. am i worng ?

DzikuVx, 2020-08-03:
Not every pin can be assigned to every timer. Refer to STM32F411 datasheet for exact mapping

adriancodre, 2020-08-03:
> Not every pin can be assigned to every timer. Refer to STM32F411 datasheet for exact mapping

thanks.  then i will put both ailerons on one servo output an use the free servor output to control tilt for the camera.
also will try adding a 16 Channel 12-bit PWM/Servo Driver-I2C interface PCA9685 module if i want to use flaperons on the future