iNavFlight/inav #6013: Platform/Mixer: how to setup a "Bicopter"?

emtrax-ltd, 2020-08-02:
How can i setup a "Bicopter" (motor left/right with servo tilt on both sides)?

When "Multicopter" or "Tricopter" is selected as type of platform, then only a single servo output is available and is mapped to S6. However, I need two servo outputs somewhere between S3 and S6. I am using a F411-WSE as flight controller.

PS: this mixer/output setup "works great" as flying wing, but will not hover or flight with the implied piff controller.

DzikuVx, 2020-08-03:
For BiCopter use Multirotor and assign 2 servos. But F411-WSE does not have outputs to handle such configuration. So not doable with this hardware

closed 2020-08-03 by DzikuVx

emtrax-ltd, 2020-08-03:
So i've one more question to clarify this: why the f411-wse does not have outputs to handle this? the controller has fully accesibale (logical/physical) outputs for 2 motors (s1, s2) and 4 seperate servos (s3, s4, s5 s6). Is the limitation you called "not doable with this hardware" implied by the definition within the target "MATEKF411SE" or the target-mixer-logic of INAV ? finally,... in Betaflight it works but INAV would be much nicer ;)

digitalentity, 2020-08-03:
You can't arbitrary map motors and servos to outputs due to internal CPU timer mapping. INAV supports two mappings for this board - one for multirotors, with 5 motors + 1 servo, one for airplanes, with 2 motors + 4 servos. They are not interchangeable at the moment, INAV will use the one, defined by mixer type. This is  something we can look into as a feature request, but not in the nearest future.

emtrax-ltd, 2020-08-03:
Thanks for the clarification. So now the next "final question": if i understood it corretly, the only thing i have to do is to patch (for myself) the bitmask of the output-type of S5 to "TIM_USE_MC_SERVO" (Servo) for "MC" (Multicopter)(src/main/target/MATEKF411SE/target.h)?

```
const timerHardware_t timerHardware[] = {
    DEF_TIM(TIM3, CH1, PB4,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S1  D(1,4,5)
    DEF_TIM(TIM3, CH2, PB5,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S2  D(1,5,5)
    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 1), // S3  D(2,1,6)
    DEF_TIM(TIM1, CH2, PA9,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 1), // S4  D(2,2,6)
    DEF_TIM(TIM1, CH3, PA10, TIM_USE_MC_SERVO | TIM_USE_FW_SERVO,  0, 1), // S5  D(2,6,6)
    DEF_TIM(TIM4, CH3, PB8,  TIM_USE_MC_SERVO | TIM_USE_FW_SERVO,  0, 0), // S6  D(1,7,2)
    ....
};
```

digitalentity, 2020-08-03:
This won't  work. Same timer can't be used for both motor and servo. You'll likely fry your servo or your motor protocol will fall back to classic 50Hz PWM:

```
DEF_TIM(TIM1, CH2, PA9,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 1), // S4  D(2,2,6)
DEF_TIM(TIM1, CH3, PA10, TIM_USE_MC_SERVO | TIM_USE_FW_SERVO,  0, 1), // S5  D(2,6,6)
```

This will work (2 motor + 4 servos):

```
    DEF_TIM(TIM3, CH1, PB4,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S1  D(1,4,5)
    DEF_TIM(TIM3, CH2, PB5,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S2  D(1,5,5)
    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_MC_SERVO | TIM_USE_FW_SERVO,  0, 1), // S3  D(2,1,6)
    DEF_TIM(TIM1, CH2, PA9,  TIM_USE_MC_SERVO | TIM_USE_FW_SERVO,  0, 1), // S4  D(2,2,6)
    DEF_TIM(TIM1, CH3, PA10, TIM_USE_MC_SERVO | TIM_USE_FW_SERVO,  0, 1), // S5  D(2,6,6)
    DEF_TIM(TIM4, CH3, PB8,  TIM_USE_MC_SERVO | TIM_USE_FW_SERVO,  0, 0), // S6  D(1,7,2)
```

emtrax-ltd, 2020-08-03:
muchas gracias - i'll give it a try

emtrax-ltd, 2020-08-03:
ok... after a quick hack, make and flash.... it works! motors are spinning and servos are moving - thank you all!

[image]
[image]

emtrax-ltd, 2020-08-03:
OK... one last comment for the case: after al little bit of confusion about weights, directions/reversions within smix and servo-output-> NOW the lady hovers with a lot of manual interventions on the sticks and flies short distances very unstable yet. so now it's time for the pid-tuning.

```
# mmix
mmix 0  1.000  1.000  0.000  0.000
mmix 1  1.000 -1.000  0.000  0.000

# smix
smix 0 4 2 -100 0 -1
smix 1 4 1 100 0 -1
smix 2 5 2 100 0 -1
smix 3 5 1 100 0 -1

# servo
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 -100
```

Greetings from the workbench

[image]

manoj453, 2023-02-28:
I am trying to build a bi-copter using an omnibus f4 pro v2 board. I am not able to apply the patch you have mentioned. can you please provide the detailed steps where to add in target.h file and how to create a hex file out of it.
my config is:
const timerHardware_t timerHardware[] = {
	DEF_TIM(TIM3, CH1, B00,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S1  D(1,4,5)
	DEF_TIM(TIM3, CH2, B01,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S2  D(1,5,5)
	DEF_TIM(TIM1, CH1, A03,  TIM_USE_MC_SERVO | TIM_USE_FW_SERVO,  0, 1), // S3  D(2,1,6)
	DEF_TIM(TIM1, CH2, A02,  TIM_USE_MC_SERVO | TIM_USE_FW_SERVO,  0, 1), // S4  D(2,2,6)
	};

emtrax-ltd, 2023-02-28:
> d how to c

You can use S5 & S6... because it is already defined for OMNIBUSF4PRO - no special fix is needed

https://github.com/iNavFlight/inav/blob/master/src/main/target/OMNIBUSF4/target.c

#if (defined(**### OMNIBUSF4PRO**) || defined(OMNIBUSF4V3)) && !(defined(OMNIBUSF4PRO_LEDSTRIPM5) || defined(OMNIBUSF4V3_S6_SS) || defined(OMNIBUSF4V3_S5S6_SS) || defined(OMNIBUSF4V3_S5_S6_2SS))
    DEF_TIM(TIM5,  CH2, PA1,  TIM_USE_MC_MOTOR | **### TIM_USE_MC_SERVO** | TIM_USE_FW_SERVO,                0, 0), // **### S5_OUT**
    DEF_TIM(TIM1,  CH1, PA8,  TIM_USE_MC_MOTOR | **### TIM_USE_MC_SERVO** | TIM_USE_FW_SERVO,                0, 0), // **### S6_OUT**

manoj453, 2023-03-01:
But I am not able to see bicopter option! 
can you please tell me how to make a bicopter with omnibus f4 pro v2 board using inav

emtrax-ltd, 2023-03-01:
Select "Multicopter"/"QuadX", remove motor S3&S4, add Server to S5&S6. 

Setup "Mixer" via CLI

# mmix
mmix 0  1.000  1.000  0.000  0.000
mmix 1  1.000 -1.000  0.000  0.000

# smix
smix 0 4 2 -100 0 -1
smix 1 4 1 100 0 -1
smix 2 5 2 100 0 -1
smix 3 5 1 100 0 -1

# servo
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 -100

manoj453, 2023-04-03:
i tried configuring the above config but my motors are not working after
giving the throttle.
I am using fs-skyi6 transmitter with self centric throttle.
i have configured swa5 as arming switch. after arming also the motors are
not able to connect.
*Regards,*
*ManojKumar*
*ph.no:8500282800 <http://ph.no:8500282800>*

manoj453, 2023-04-04:
[image]
Above is the connection diagram i am using

manoj453, 2023-04-04:
[image]
The configuration you have mentioned

manoj453, 2023-04-07:
Since my servos are enabled on s5 and s6 I updated the data as below and tried. Moters and servos are working now but not in sync.

while giving the throttle the servos are moving in opp direction
# mmix
mmix 0 1.000 1.000 0.000 0.000
mmix 1 1.000 -1.000 0.000 0.000

# smix
smix 0 5 2 -100 0 -1
smix 1 5 1 100 0 -1
smix 2 6 2 100 0 -1
smix 3 6 1 100 0 -1

# servo
servo 5 1000 2000 1500 -100
servo 6 1000 2000 1500 100

manoj453, 2023-04-13:
when ever i give the throttle the channel 5 goes to zero