iNavFlight/inav #6014: Unable to flash matekf411se - disconnects and loses port

JohnHennesey, 2020-08-03:
New to inav, take it easy on me if asking something dumb.

I have a new Matek F 411 SE flight controller - firmware is from February, would like to update it to the latest.  I am able to connect in the stand alone iNav configurator via com4, 57600 baud.  Can make changes, save, reboot, etc.

When I try to flash with the latest it makes the windows disconnect 

## Steps to Reproduce
1. Connect FC to the computer
2. Connect to FC - in my case com 4, 57600
3. Click Disconnect
4. Click Firmware flasher tab
5. Select MATEKF411SE, 2.5.1 - MATEK411SE - 2020-6-16 12:32 (stable)
5.1 Enable 'Full chip erase'
5.2 Select Manual baud rate - 57600
6. Click Load firmware (online)
7. Click 'Flash Firmware'
8. Hear windows disconnect sound
9. Logs say 'Failed to flash COM4'
10. COM4 is no longer a choice to connect to the FC, have to restart iNav to connect.

## Expected behavior
Able to flash newer version to FC.

## Additional context
The logs:
CONFIGURATOR2.5.0
Connect
Wireless mode 
Hide Log
2020-08-02 @ 22:02:55 -- Running - OS: Windows, Chrome: 78.0.3904.87, Configurator: 2.5.0
2020-08-02 @ 22:02:55 -- Loaded release information from GitHub.
2020-08-02 @ 22:03:04 -- Serial port successfully opened with ID: 1
2020-08-02 @ 22:03:04 -- MultiWii API version received - 2.4.0
2020-08-02 @ 22:03:04 -- Flight controller info, identifier: INAV, version: 2.4.0
2020-08-02 @ 22:03:04 -- Running firmware released on: Feb 11 2020 23:07:15
2020-08-02 @ 22:03:04 -- Board: M41S, version: 0
2020-08-02 @ 22:03:04 -- Unique device ID received - 0x3[removed]
2020-08-02 @ 22:03:05 -- PWM output is disabled. Motors and servos will not work. Use Outputs tab to enable!
2020-08-02 @ 22:03:10 -- Serial port successfully closed
2020-08-02 @ 22:05:18 -- Failed to flash COM4

---

- FC Board name and vendor: MATEKSYS F411-WSE
- INAV version string:

##Version info (by entering 'version' in cli):
version
INAV/MATEKF411SE 2.4.0 Feb 11 2020 / 23:07:15 (bcd4caef9)
GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

DzikuVx, 2020-08-03:
Flashing via COM4 is not supported by hardware. Only USB and COM1 in some cases

closed 2020-08-03 by DzikuVx

JohnHennesey, 2020-08-03:
Thank you DzikuVx.  For the next (newbie) guy who finds this, putting the FC into DFU mode allowed me to flash it.

[This ](https://www.youtube.com/watch?v=0mrJG7AOmRI)video was useful in showing me how to update the drivers and connect in DFU mode.   Had to reboot afterwards to get iNav to connect via COM4.  I'm sure it was user error.  Helpful part starts around 12 minutes in.

difusao, 2021-08-12:
I sucessfull in video https://www.youtube.com/watch?v=0mrJG7AOmRI
impulseRC driver fixer - https://impulserc.com/pages/downloads

Mangjana, 2023-08-04:
I Have almost the same Problem, i Using MAC, my flight controller Speedybee F405 Wing. Before the Firmware was ardupilot(but I don't like Ardupilot, to complicated for newbee), so I decided to go back to INAV. Everything was good. But after flashing firm ware, And Connected for view minute only and disconnect.
I Could't see anymore "/dev/cu.usbmodem0x0
[image]
000001" (mac) or com1 in win
Please help me @DzikuVx and other friend
[image]