iNavFlight/inav #6015: Position Hold issue

alexjoseph3141, 2020-08-03:
## Current Behavior
I am having a sort of toilet bowl effect with POS HOLD mode on my quadcopter but it is relatively small and doesn't get out of control. I calibrated my magnetometer and it is far from any interference. I was told it was a problem with the mag but other modes like alt hold work fine. I am new to this hobby and can't find the problem in the log so if someone could help that would be great. Also, the heading I see in the osd is right so I don't think its the mag but I am not sure.

[blackbox_log_2020-08-03_012616.TXT](https://github.com/iNavFlight/inav/files/5014159/blackbox_log_2020-08-03_012616.TXT)

## Steps to Reproduce
1. Turn on POS-HOLD

## Expected behavior
I just expected the quad to hover in position but it moved more then I expected.

## Suggested solution(s)
Not really sure, I'm trying to find a solution

## Additional context
# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# logic

# gvar

# gf

# feature
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 0 64 115200 115200 0 115200
serial 5 2 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100
aux 1 1 5 900 2100
aux 2 3 5 1700 2100
aux 3 11 6 1700 2100
aux 4 46 4 1700 2100
aux 5 13 2 1700 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 2 6 V
osd_layout 0 1 0 11 V
osd_layout 0 2 0 0 V
osd_layout 0 4 8 6 H
osd_layout 0 6 23 9 V
osd_layout 0 7 1 4 V
osd_layout 0 9 1 2 H
osd_layout 0 10 10 14 H
osd_layout 0 13 25 4 H
osd_layout 0 14 1 5 V
osd_layout 0 15 25 7 V
osd_layout 0 20 20 10 H
osd_layout 0 21 20 9 H
osd_layout 0 22 23 10 V
osd_layout 0 23 24 4 V
osd_layout 0 24 12 4 V
osd_layout 0 28 23 11 H
osd_layout 0 30 0 12 V
osd_layout 0 33 1 2 H
osd_layout 0 34 10 3 H
osd_layout 0 38 1 10 V
osd_layout 0 48 23 10 H
osd_layout 0 55 1 8 V
osd_layout 0 96 25 6 V

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 110
set gyro_lpf_type = PT1
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 120
set acc_hardware = MPU6000
set acczero_x = 6
set acczero_y = 7
set acczero_z = -93
set accgain_x = 4130
set accgain_y = 4114
set accgain_z = 4075
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set mag_declination = -1230
set magzero_x = -16
set magzero_y = -299
set magzero_z = -373
set mag_calibration_time = 40
set baro_hardware = BMP280
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SBUS
set blackbox_rate_denom = 2
set motor_pwm_rate = 16000
set motor_pwm_protocol = DSHOT600
set throttle_idle =  7.000
set failsafe_procedure = DROP
set align_board_roll = -4
set align_board_pitch = -14
set model_preview_type = 3
set small_angle = 40
set applied_defaults = 2
set gps_min_sats = 5
set mc_airmode_type = THROTTLE_THRESHOLD
set inav_auto_mag_decl = OFF
set inav_w_z_baro_p =  0.200
set inav_w_z_gps_p =  0.350
set inav_w_z_gps_v =  0.110
set nav_auto_speed = 100
set nav_auto_climb_rate = 200
set nav_manual_speed = 250
set nav_manual_climb_rate = 98
set nav_landing_speed = 214
set nav_emerg_landing_speed = 300
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 500
set nav_mc_bank_angle = 45
set nav_mc_hover_thr = 1421
set osd_units = IMPERIAL
set osd_main_voltage_decimals = 2

# profile
profile 2

# battery_profile
battery_profile 1

set bat_cells = 4
set vbat_cell_detect_voltage = 440
set vbat_max_cell_voltage = 425
set vbat_min_cell_voltage = 333
set vbat_warning_cell_voltage = 346
set battery_capacity = 1550

# end the command batch
batch end

---

# INAV/OMNIBUSF4 2.5.1 Jun 16 2020 / 12:28:02 (098a574b3)
# GCC-9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

renanseibert, 2020-09-15:
@alexjoseph3141 
Hi
Were you able to solve the problem? I have a similar problem, I still couldn't get the quad to set to POSHOLD, but it was more stable when increasing the HOVER THROTTLE.
Did you find another solution?

alexjoseph3141, 2020-09-16:
Nope. Sorry

renanseibert, 2020-09-16:
@alexjoseph3141 

Hi,
Today I managed to solve the problem of stabilizing in POSHOLD mode. It was necessary to correct the navigation PID value. The D gain was too high and therefore the motors err very aggressively when correcting the xy position. POSHOLD now works perfectly. (Sorry for my english ...)

alexjoseph3141, 2020-09-23:
Thank you so much, and your English is great! I wouldn't have noticed until
you said something.

closed 2020-12-11 by DzikuVx

Palput, 2021-03-16:
ill try it, thanks

SJurick, 2023-04-05:
Just in case this helps someone.  This is probably in the documentation "somewhere" but I could not get PHOLD to work worth a crap.  Tried calibrating numerous times out in middle of backyard.  Tried PID tuning (got worse).  Re-soldered my connections to GPS.  Quad would not sit still.  Then I looked closely at the Magnetometer orientation I had set in iNAV.  I used my mouse to rotate the quad and GPS unit's graphic on the screen and realized the picture of the GPS had a plug.  Mine was sitting sideways.  I turned it to be the same as the real thing and poof - quad holds position.  This is a bit misleading at least for me but I was orientating the GPS based on the little dot on top of the GPS which on mine is off center a tad.  Learned a lesson.  Hope this helps someone else!