iNavFlight/inav #6016: Quadcopter slowly spins around YAW axis when in GPS Poshold mode

DiscoMan18, 2020-08-03:
It just spins around YAW Axis when the quad hovers in place in GPS Poshold mode. The heading drifts with the rate of about 1 degree per minute, so in 20 minutes it drifts by 20 degrees. My YAW Pids are 70, 45, 0. Should I raise the I term to fix that behaviour? or nav_mc_heading_p should be increased too? By default it's set to 60.

digitalentity, 2020-08-04:
Verify that your sticks are centered at 1500 in the receiver tab.

DiscoMan18, 2020-08-04:
> Verify that your sticks are centered at 1500 in the receiver tab.

Already checked it, yaw, roll and pitch channels are at 1500 when I don't touch the sticks. I already solved this problem by enabling HEADING HOLD mode along with GPS POSHOLD. Didn't you think on joining this two modes together under GPS POSHOLD? As to me, it seems logical. This way the quad will keep it's 3D orientation including heading when there is no user input.

closed 2020-09-09 by DzikuVx