iNavFlight/inav #6017: Build error

fenglaijun, 2020-08-04:
i had met this error,please help me,thanks!

make SPRACINGF3
echo "" && \
        echo "Building SPRACINGF3" && \
        CFLAGS= make -j 8 TARGET=SPRACINGF3 && \
        echo "Building SPRACINGF3 succeeded."

Building SPRACINGF3
make[1]: 正在进入目录 `/home/dog/inav'
/home/dog/inav/src/utils/compiler.rb:38: warning: Insecure world writable dir /home/dog in PATH, mode 040777
settings.yaml -> settings_generated.h, settings_generated.c
./src/utils/settings.rb:748: syntax error, unexpected tLPAREN_ARG, expecting keyword_do_LAMBDA or tLAMBEG
        add_condition = -> (c) {
                            ^
./src/utils/settings.rb:748: syntax error, unexpected tLAMBEG, expecting keyword_end
./src/utils/settings.rb:755: syntax error, unexpected '}', expecting keyword_end
./src/utils/settings.rb:887: syntax error, unexpected tLPAREN_ARG, expecting keyword_do_LAMBDA or tLAMBEG
        add_value = -> (v) {
                        ^
./src/utils/settings.rb:887: syntax error, unexpected tLAMBEG, expecting keyword_end
./src/utils/settings.rb:894: syntax error, unexpected '}', expecting keyword_end
make[1]: *** [obj/main/SPRACINGF3/settings_generated.h] 错误 1
make[1]:正在离开目录 `/home/dog/inav'
make: *** [SPRACINGF3] 错误 2

shellixyz, 2020-08-04:
Your version of Ruby is probably too old. What is `ruby -v` displaying ?

fenglaijun, 2020-08-04:
> Your version of Ruby is probably too old. What is `ruby -v` displaying ?

thanks!this is because i did not intall ruby.now i can build it success!

fenglaijun, 2020-08-04:
ruby didn't install

closed 2020-08-04 by fenglaijun