iNavFlight/inav #6019: Modes displayed in osd

juiceman83, 2020-08-04:
?I'm new to useing  inav , I  set up all flight modes correctly but I was wondering why I can't see modes show up my OSD like ,auto tune, servo trim,  rth ,while I'm on the bench, by the way I'm useing a wing ,does the wing have to be airborne for modes to be displayed.

shellixyz, 2020-08-04:
Only the mode which is actually engaged will show up on the OSD and you can't engage the modes you mentioned without arming the flight controller first.

juiceman83, 2020-08-05:
> Only the mode which is actually engaged will show up on the OSD and you can't engage the modes you mentioned without arming the flight controller first.

Thanks for your help sir,but the board was armed with six sats,I just was wondering why I could see angle and horizon mode , manual mode but no auto tune or servo trim ,rth mode,to let know if these modes are setup right and active

Pairan, 2020-08-05:
You could check it via the Configurator- connect the plane, switch to modes tab and watch what happens, when you move your sticks to enter RTH or AUTO TUNE ... the slider should walk into the right position and the Field to the left (with the mode on it) should be lit blue.

Hope this helps!

juiceman83, 2020-08-05:
Thanks so much

closed 2020-08-14 by DzikuVx